TheThingsNetwork/lorawan-stack #3830: Add cluster hints to Applications

htdvisser, 2021-02-19:
#### Summary

I think it would be helpful to add cluster hints to applications.

#### What is already there? What do you see now?

With explicit linking gone, Applications in v3 are no longer explicitly linked (between Application Server and Network Server) in a specific cluster. This means that Applications can now contain devices in different clusters. This unfortunately also makes it easier to make mistakes, for example by registering some devices in `eu1`, and some devices in `nam1`. I've solved this by creating different applications for different clusters:

[image]

#### What is missing? What do you want to see?

In such a setup, it may be useful to set the `network_server_address`, `application_server_address` and `join_server_address` in the Application, so that the Console can give hints to inform the user that an application is in a different cluster.

#### Environment

The Things Network, The Things Stack Cloud

#### How do you propose to implement this?

- Add those fields to the Application database model
- Add those fields to the Application API
- Make the Console ask for those fields when registering an application (they are optional)
- Make the Console display hints when listing / viewing applications

#### Can you do this yourself and submit a Pull Request?

Let's discuss this first, maybe @kschiffer has some useful UX insights before we just start implementing this.

kschiffer, 2021-12-20:
We can implement this complementary to #4977. For this we'd need to extend the application model though to include  `network_server_address`, `application_server_address` and `join_server_address` as pointed out by @htdvisser. Then we can check those and hide such applications from the list, when all of the servers point to different clusters.

@htdvisser can you make the start with extending the application model or delegate this accordingly?

htdvisser, 2022-01-24:
API and database models completed in https://github.com/TheThingsNetwork/lorawan-stack/compare/issue/3830-app-server-address. 

@ryaplots you can continue working on that branch. Let me know if you need anything else.

htdvisser, 2022-02-16:
Since this unfortunately didn't make it into v3.18.0, and this requires a database migration, we have to postpone it until the v3.19.0 release in April.

kschiffer, 2022-03-10:
Does anything speak against making this `network_server_address`, `application_server_address` and `join_server_address` assignment implicit in the creation form? Meaning that these fields will be set automatically without allowing any modification.

I feel like adding these fields explicitly to the application form will make it more confusing (even when hiding it in a advanced settings section). Also, I don't see any use case where a user would want to create an application for a different cluster.

htdvisser, 2022-03-10:
No, that's totally fine.

ryaplots, 2022-03-11:
> API and database models completed in [`issue/3830-app-server-address`](https://github.com/TheThingsNetwork/lorawan-stack/compare/issue/3830-app-server-address).
> 
> @ryaplots you can continue working on that branch. Let me know if you need anything else.

I pushed my work to the branch already

htdvisser, 2022-03-24:
I'll rebase this and open a pull request for v3.19 after v3.18.2 is out.

PR #5337 (2022-04-12): Add server addresses to Application

#### Summary

This pull request adds NS/AS/JS addresses to the Application model and API, updates the CLI to warn about applications registered to another cluster than the one currently configured. @ryaplots also updated the Console in a similar way.

Closes #3830

#### Changes

- Add `{network,application,join}_server_address` to Application model and API
- Make IS refuse end device registrations in clusters other than the one set by the Application
- Make CLI and Console set application addresses and complain if they don't match the configuration

#### Te [... cut]
closed 2022-04-12 by htdvisser