TheThingsNetwork/lorawan-stack #3833: Add UsedAt field to ContactInfoValidation

htdvisser, 2021-02-19:
#### Summary

It would be helpful to add a `used_at` field to the `ContactInfoValidation` model so that we can tell the difference between "token not found" and "token already used".

#### What is already there? What do you see now?

Right now users are confused, because the Account app (?) tells them that "the validation token was not found" and that they should "contact an administrator".

Refs https://github.com/TheThingsIndustries/lorawan-stack-support/issues/318

#### What is missing? What do you want to see?

We should show users a different error message depending on whether the validation token was not found or already used.

#### How do you propose to implement this?

- Add a `used_at` field to the database model
- Check that before validating the contact info in `contactInfoStore.Validate`
- Set it after validating the contact info in `contactInfoStore.Validate`
- Maybe update the console, as this seems to be a custom error message (i.e. the Console doesn't just forward the error from the backend).

#### How do you propose to test this?

Unit tests

#### Can you do this yourself and submit a Pull Request?

For the Identity Server part, yes, for the Console, no.

htdvisser, 2021-03-08:
Identity Server changes are in https://github.com/TheThingsNetwork/lorawan-stack/compare/feature/3833-validation-token-used. Handing over Console part to @kschiffer.

asmulko, 2021-03-22:
Looks like console handles this error correctly @htdvisser 

[image]

PR #3980 (2021-03-29): Change error for already used contact info validation token

#### Summary

This pull request changes the error that the IS returns when consuming a contact info validation token that was already used.

Closes #3833.

#### Changes

- Add column to DB model
- Introduce new error message

#### Testing

- Unit test
- 🐒 
- 🐒 https://github.com/TheThingsNetwork/lorawan-stack/issues/3833#issuecomment-804156980

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configurat [... cut]
closed 2021-03-29 by htdvisser