TheThingsNetwork/lorawan-stack #3838: Token refresh refused when leaving console open for long time

jpmeijers, 2021-02-21:
#### Summary

[image]

#### Steps to Reproduce

1. Log into the console
2. Go to a device's details page
3. Leave this tab open for a couple of hours

#### What do you see now?

The screenshot above, after which I need to refresh the page to see something again.

#### What do you want to see instead?

Don't log me out while I have the tab open.

#### Environment

TTN v3.11.0

#### How do you propose to implement this?

...

#### How do you propose to test this?

Open a tab and leave it open for a day.

#### Can you do this yourself and submit a Pull Request?

No

jpmeijers, 2021-02-27:
I'm not sure if this is realted, but I had both TTN v3 and a private V3 instance open in my browser since about the same time yesterday afternoon. This morning both showed the same error screen:

[image]

Logs from the private stack attached in case it's useful.
[stack-logs-202002271005.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6053939/stack-logs-202002271005.txt)

johanstokking, 2021-07-13:
This looks pretty stale, let's triage

johanstokking, 2021-07-20:
Please reopen if this issue still exists.

closed 2021-07-20 by johanstokking