TheThingsNetwork/lorawan-stack #3842: Cloud Formation stack lambda function errors

thinkOfaNumber, 2021-02-23:
Hi,

I think this is a bug, but please correct me if I'm wrong!

#### Summary

After installing and configuring the AWS IoT integration for The Things Stack in our AWS account with the cluster address `au1.cloud.thethings.network`, I created a thing in AWS IoT Core. The CreateThingFunction returned an error: `Failed to dial Identity Server at "ttn-v3-is:///ttn.au1.cloud.thethings.network": connection error: desc = "transport: error while dialing: dial tcp: lookup ttn.au1.cloud.thethings.network on 169.254.78.1:53: no such host"` The documentation says that while the cluster address can be in au1, `Identity Server API requests to any cluster other than eu1 will fail`.

I changed the stack so that the cluster address was `eu1.cloud.thethings.industries` and it seemed to work in that things were created on ttn console au1.cloud.thethings.network, but the lambda function is still reporting failure. The logs show 
```
2021/02/23 02:13:20 Created on Identity Server
2021/02/23 02:13:21 Failed to dial Join Server at "ttn-v3-js:///eu1.cloud.thethings.network": context deadline exceeded
2021/02/23 02:13:21 Rollback create on Identity Server failed: rpc error: code = DeadlineExceeded desc = context deadline exceeded
```

#### Steps to Reproduce

1. create an organisation and application on au1.cloud.thethings.network using the au1 community network
2. install the AWS IoT Integration 1.0.4 for The Things Stack, using an au1 cluster address (no tenant ID)
3. create a thing in AWS

#### What do you see now?

The cluster address given in the documentation doesn't seem to work, or isn't applicable for community network. It appears eu1 has to be used instead. The lambda function reports an error result in cloudwatch, but creates things in the TTN console.

#### What do you want to see instead?

The lambda function shouldn't report an error. The documentation [here ](https://www.thethingsindustries.com/docs/integrations/aws-iot/default/deployment-guide/) should have a sample cluster address for the community network.

#### Environment

I don't think the hardware or gateways are relevant here.

#### How do you propose to implement this?

Documentation updates and lambda function updates.

#### How do you propose to test this?

Could easily be tested by using the reproduce steps above so long as eu1 is not used as the test network.

#### Can you do this yourself and submit a Pull Request?

Unfortunately not if the lambda functions are written in Go!

Thanks

johanstokking, 2021-02-23:
Thanks @thinkOfaNumber, we have TTN V3 support for the AWS IoT Core Integration ready for deployment. We're currently waiting for our The Things Stack Cloud to be upgraded to 3.11.x so we can start the roll-out.

Internal reference https://github.com/TheThingsIndustries/lorawan-stack-aws-integration/pull/21

thinkOfaNumber, 2021-02-23:
Thanks for the quick response, I didn't realise this wasn't ready now. Is there a roadmap somewhere I can refer to?

johanstokking, 2021-02-24:
References also https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/236

johanstokking, 2021-02-24:
This is now released, please see https://www.thethingsindustries.com/docs/integrations/aws-iot/default/deployment-guide/ and be sure to select **Community** before clicking the CloudFormation deployment button.

closed 2021-02-24 by johanstokking