TheThingsNetwork/lorawan-stack #3864: Handling Non-Compliant End Devices

htdvisser, 2021-02-25:
#### Summary

We need to think about how we're going to handle non-compliant end devices.

#### What is already there? What do you see now?

With more users migrating to v3, we see an increasing number of non-compliant (or misconfigured) end devices. While most of these end devices are "hobby basement" devices that only impact the coverage area of the developers own gateway, we can expect some of these devices to eventually end up in places where they could do more damage. 

The most common issues are:

- End devices (ABP) that use DevAddrs of a different network
- End devices (ABP) with frame counter resets enabled
- End devices (OTAA) with join nonce resets enabled
- End devices that use fixed data rates (such as SF12BW125 in EU868)
- End devices (ABP) that don't use all available channels
- End devices that always use confirmed uplinks
- End devices not respecting the EU868 duty cycle
- End devices not respecting the Join Request duty cycle
- End devices not respecting the Fair Access Policy of The Things Network
	- 30 seconds of uplink airtime per day
	- 10 downlinks per day (application downlinks and ACKs)
- End devices not respecting MAC commands

#### What is missing? What do you want to see?

We need to think about how we can limit the impact of such devices, and at the same time about helping developers address these issues before the devices end up in production.

We can start by making the Network Server set some flags in the EndDevice registry when it detects that the end device is showing non-compliant or unexpected behavior.

When an end device is repeatedly / continuously in violation we can trigger warnings, disable functionality, or in the worst case completely ban the device.

#### How do you propose to implement this?

I propose that the Network Server keeps a set of flags/counters that are set/incremented when an end device violates a guideline. At some threshold we can then start taking actions such as:

- trigger warning events
- showing warnings on the device page of the Console
- showing warnings in the device list of the Console
- disable Network Server functionality for the device
	- stop sending MAC commands that are not respected
	- stop sending downlinks altogether
	- stop delivering uplinks to the application

#### How do you propose to test this?

Not sure. This is going to be pretty difficult to test.

#### Can you do this yourself and submit a Pull Request?

Let's first discuss this.

kschiffer, 2021-03-04:
I immediately thought of a "Issues" tab on both the application and end device layer, where such issues are listed. The tab would have a small counter bubble that should make it easy to spot issues when on the device/app overview. 

I understand that we can categorize these issues into severities, meaning that some will issues warning only while other will have consequences as listed above.

johanstokking, 2021-03-23:
I think that we should develop this as event consumers for the Event Server (https://github.com/TheThingsIndustries/lorawan-stack/issues/1804). Aim for implementing this by observing events, and not include this complexity and state to an already complex Network Server. Should we emit additional event payload or new events, then that's of course fine.

As with other use case for observing events (gateway lifeline, channel utilization over time, etc), we need to aggregate events and keep state per entity.

meten-natuurlijk, 2021-05-29:
I fully understand the need/urge to label/sticker nodes based on their behaviour.
I also think it greatly would help if users could see in the TTN console if their nodes behave nicely within the mentioned boundaries. (airtime, etc)
I believe these node-labels could be categorized like "info", "notice", "warning" or "error" to help developers/user change the behaviour of their nodes.

meten-natuurlijk, 2021-05-29:
I believe we should not auto-exclude nodes from the network.
My personal use-case:
I spend a lot of time mapping the coverage of gateways using TTN-Mapper with a node that has a push-button.
It would take a lot more time to map the coverage of a gateway if I could only send an uplink every 5 minutes.

KrishnaIyer, 2024-10-22:
A lot of the cases here are enforced using https://www.thethingsindustries.com/document/sla/ on The Things Stack Cloud.

The rules are a bit looser for The Things Stack Sandbox since that deployment is meant for experimentation. And we have metrics and alerts to identify large spammers.

For The Things Stack Open source and The Things Stack Enterprise deployments, the operators can define themselves how they manage these cases.

This issue can be closed.

closed 2024-10-22 by KrishnaIyer