TheThingsNetwork/lorawan-stack #3874: First f_cnt is empty/null

descartes, 2021-03-01:
#### Summary
The first f_cnt value after join, ie a value that could or should be 0, is blank. If it could be populated with 0 it will save on managing database integrity issues.

#### Steps to Reproduce
Add a new device, trigger an uplink, look at the Event details, it's not there for the first uplink.

Doesn't appear in Data Storage or in WebHook.

#### What do you see now?
```
"uplink_message": {
      "session_key_id": "AXft26PUOd3Eai4jjy/JdA==",
      "f_port": 1,
      "frm_payload": "IA5BCkYToB26",
      "decoded_payload": {
        "Humidity": 50.24,
        "Pressure": 76.1,
        "Temperature": 26.3,
        "battery": 3.649,
        "flags": 32
      },
```
#### What do you want to see instead?
```
"uplink_message": {
      "session_key_id": "AXft26PUOd3Eai4jjy/JdA==",
      "f_port": 1,
      "f_cnt": 0,                      <----- this
      "frm_payload": "IA5BCkYToB26",
      "decoded_payload": {
        "Humidity": 50.24,
        "Pressure": 76.1,
        "Temperature": 26.3,
        "battery": 3.649,
        "flags": 32
      },
```

#### Environment
n/a

#### How do you propose to implement this?
I could dig in to the Go code but ...

#### How do you propose to test this?
Cycle the power on a device, see what happens.

#### Can you do this yourself and submit a Pull Request?
No!

HeadBoffin, 2021-03-12:
We run the risk of technical debt in thousands of code fragments across the world if this isn't resolved - or at least acknowledgement that that is where we are heading. Please.

johanstokking, 2021-03-16:
All fields that have the default value aren't included. So in this case it's in an integer and the value is `0`, so it's omitted.

An underlying library is doing the JSON encoding, from which we are moving away. That's a big refactor that we've been planning for a long time, see #2798.

closed 2021-03-16 by johanstokking

matthijskooijman, 2021-05-20:
Hm, I also ran into this. @johanstokking you closed this, which suggests you're saying this works as intended, but your comment partly suggests that you're planning to fix it. If this is indeed considered a bug, wouldn't it be better to keep this issue open?

Regarding omitting default values, I believe that that is indeed something specified by the protobuf protocol that I guess is used internally, but this is not normally done for JSON, which is where this becomes visible. AFAIU protobuf always works with a completely known schema of a message, so the fact that default values are omitted is really an encoding detail, when a protobuf message is decoded, these missing values can automatically be resolved to their default values, and the fact that the values were missing in the encoded protobuf shouldn't really be visible in the decoded data. But maybe you're saying that the protobuf library is also handling the JSON encoding and is leaking the omission of default values into the JSON, and then the direction #2798 is headed in will probably implicitly fix this issue as well?

matthijskooijman, 2021-06-01:
> All fields that have the default value aren't included. So in this case it's in an integer and the value is 0, so it's omitted.

I just realized that this applies more broadly, i.e. a packet with an SNR of 0 would also have the SNR field omitted, which makes it really ambiguous whether no SNR is known somehow, or the value is actually 0...

floodnetwork, 2021-11-12:
I am none the wiser from this thread. All I know is that the first f_cnt is missing and I had to go and look for it to discover it's implicitly zero. Googling and reading github is hard to turn into code. Explicit zeros are not exactly a big ask.

johanstokking, 2021-11-12:
I agree @floodnetwork. We plan to address this, but first we need to control how JSON is marshaled.

We were using standard JSON marshaling for Protocol Buffers, which omit empty values, and 0 is empty so it's omitted. That is a undesired side effect from an early V3 design decision that we are reverting now.