TheThingsNetwork/lorawan-stack #3878: MQTT application server disconnect after one hour

arigon, 2021-03-02:
#### Summary
When using the MQTT server of the new V3 Application Server (public) my MQTT client gets disconnected after one hour even there is data being pushed all the time.

#### Steps to Reproduce
1. Connect to the TTN MQTT Server
2. Wait one hour

#### What do you see now?
MQTT client is is disconnected

#### What do you want to see instead?
MQTT client is still connected

#### Environment
V3

#### How do you propose to implement this?
Disable the disconnect

#### How do you propose to test this?
Connect and wait one hour

#### Can you do this yourself and submit a Pull Request?
Nope

htdvisser, 2021-04-14:
Since @adriansmares is off today, and it isn't in progress yet, this isn't going to make today's deadline, so moved to v3.12.2.

adriansmares, 2021-04-15:
I've looked into this issue. It occurs when the client is not specifying any keep alive interval as part of the connect request.

https://github.com/TheThingsIndustries/mystique/blob/76715b4819349483b45c3906649b6ac4eb3e382b/pkg/session/connect.go#L102-L106

My recommendation for such cases is to specify the keep alive interval while creating the client, instead of leaving it at `0`. I'd also recommend using webhooks instead of MQTT for traffic processing.

I also believe that this one hour timeout should stay, in order to ensures that we don't leak connections. cc @johanstokking

arigon, 2021-04-15:
When I set KeepAlive to 2400 it would result in 1 hour as well, so that wouldn't make a difference. I think the problem is, that the server needs PINGREQ from the client to know the client is still connected. Is the TTN MQTT server supporting PINGREQ/PINGRESP?

adriansmares, 2021-04-15:
> Is the TTN MQTT server supporting PINGREQ/PINGRESP?

Yes . https://github.com/TheThingsIndustries/mystique/blob/76715b4819349483b45c3906649b6ac4eb3e382b/pkg/session/session.go#L207-L208

arigon, 2021-04-15:
I will try this. Thanks!

arigon, 2021-04-15:
The Paho MQTT Client is sending PINGREQ when using KeepAlive. I will close the issue.

closed 2021-04-15 by arigon