TheThingsNetwork/lorawan-stack #3882: v3 - AWS IoT integration Lambda function errors

robsonos, 2021-03-03:
Hi,

I am not sure if this is actually a bug or if I am I missing something.

#### Summary
I have been trying to use the AWS IoT integration with v3 au community cluster. I am still not able to make the integration work fully at the moment.  The issue occurs when trying to create a device from AWS IoT. 

#### Steps to Reproduce
1. I followed the [Deployment Guide](https://www.thethingsindustries.com/docs/integrations/aws-iot/default/deployment-guide/) for The Things Network (ap-southeast-2)
2. I followed the instructions from [Creating Things](https://www.thethingsindustries.com/docs/integrations/aws-iot/default/things/) and on AWS IoT core.

#### What do you see now?
The CloudWatch logs for the Create Thing lambda function is reporting the following:
```
2021-03-03T21:31:04.089+09:30 | 2021/03/03 12:01:04 Failed to create on Identity Server: rpc error: code = Unimplemented desc 
2021-03-03T21:31:04.128+09:30 | rpc error: code = Unimplemented desc = : Error null
```

#### What do you want to see instead?
A device created in both AWS IoT core and TTN v3 au console.

#### Environment
Nothing relevant to add here.

#### How do you propose to implement this?
Clarify what the problem is and update the documentation.

#### How do you propose to test this?
I could reproduce the steps above to test whether the issue has been fixed.

#### Can you do this yourself and submit a Pull Request?
Unfortunately not.

Cheers.

johanstokking, 2021-03-04:
Thanks @robsonos for filing the issue. I have a report by somebody else, but I was unable to reproduce the issue myself. It looks like you have the means to help me figure out what's going on here.

Can you please do the following:

1. Copy/paste the **Environment Variables** section of the AWS Lambda function `CreateThingFunction`? You can find this in Lambda (make sure you're in the right region). The actual function names includes some arbitrary CloudFormation hashes but you should be able to recognize the function easily. Note that one of the environment variables is your secret API key. Please blur or remove before posting here – I don't need to have access to your application

2. Add a new variable `DEBUG` and set the value to `true`, and then run again. Copy/paste the AWS CloudWatch logs of the `CreateThingFunction` with extended debug logs. Again, please strip any confidential information

johanstokking, 2021-03-04:
Nevermind, we can reproduce.

johanstokking, 2021-03-04:
@robsonos Please update your CloudFormation stack per https://www.thethingsindustries.com/docs/integrations/aws-iot/default/update/. Make sure that the version is 1.1.2.

robsonos, 2021-03-04:
HI @johanstokking,

I updated the CloudFormation stack to version 1.1.2 and the create and delete thing functions appeared to work (they did create and removed devices from my application on ttn v3 console). However, the CloudWatch logs is still reporting some errors:

Create Thing Lambda function:
```
2021-03-05T08:01:56.242+09:30	START RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84 Version: $LATEST
2021-03-05T08:01:59.242+09:30	2021/03/04 22:31:59 Failed to dial Identity Server at "ttn-v3-is:///eu1.cloud.thethings.network": context deadline exceeded
2021-03-05T08:01:59.243+09:30	END RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84
2021-03-05T08:01:59.243+09:30	2021-03-04T22:31:59.243Z 792b0d59-43ca-461a-b2cb-b8013fc37d84 Task timed out after 3.00 seconds
2021-03-05T08:02:52.431+09:30	START RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84 Version: $LATEST
2021-03-05T08:02:54.536+09:30	2021/03/04 22:32:54 Created on Identity Server
2021-03-05T08:02:55.213+09:30	2021/03/04 22:32:55 Created on Join Server
2021-03-05T08:02:55.432+09:30	2021/03/04 22:32:55 Failed to create on Network Server: rpc error: code = DeadlineExceeded desc = context deadline exceeded
2021-03-05T08:02:55.432+09:30	2021/03/04 22:32:55 Rollback create on Join Server failed: rpc error: code = DeadlineExceeded desc = context deadline exceeded
2021-03-05T08:02:55.432+09:30	2021/03/04 22:32:55 Rollback create on Identity Server failed: rpc error: code = DeadlineExceeded desc = context deadline exceeded
2021-03-05T08:02:55.435+09:30	END RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84
2021-03-05T08:02:55.435+09:30	2021-03-04T22:32:55.435Z 792b0d59-43ca-461a-b2cb-b8013fc37d84 Task timed out after 3.00 seconds
2021-03-05T08:04:43.615+09:30	START RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84 Version: $LATEST
2021-03-05T08:04:46.156+09:30	2021/03/04 22:34:46 Failed to create on Identity Server: rpc error: code = AlreadyExists desc = error:pkg/identityserver/store:id_taken (ID already taken)
2021-03-05T08:04:46.224+09:30	Put event logs failed: InvalidSequenceTokenException
2021-03-05T08:04:46.319+09:30	rpc error: code = AlreadyExists desc = error:pkg/identityserver/store:id_taken (ID already taken): Error null
2021-03-05T08:04:46.320+09:30	END RequestId: 792b0d59-43ca-461a-b2cb-b8013fc37d84
```

Delete Thing Lambda function:
```
2021-03-05T08:05:55.243+09:30	START RequestId: 0da9902a-6c6e-45be-926d-545abef5c355 Version: $LATEST
2021-03-05T08:05:58.243+09:30	2021/03/04 22:35:58 Failed to delete from Identity Server: rpc error: code = DeadlineExceeded desc = context deadline exceeded
2021-03-05T08:05:58.243+09:30	2021/03/04 22:35:58 Failed to dial Join Server at "ttn-v3-js:///au1.cloud.thethings.network": context deadline exceeded
2021-03-05T08:05:58.246+09:30	END RequestId: 0da9902a-6c6e-45be-926d-545abef5c355
2021-03-05T08:05:58.246+09:30	2021-03-04T22:35:58.246Z 0da9902a-6c6e-45be-926d-545abef5c355 Task timed out after 3.00 seconds
2021-03-05T08:07:04.433+09:30	START RequestId: 0da9902a-6c6e-45be-926d-545abef5c355 Version: $LATEST
2021-03-05T08:07:06.870+09:30	2021/03/04 22:37:06 Failed to get device: rpc error: code = NotFound desc = error:pkg/identityserver/store:end_device_not_found (end device `******` not found)
2021-03-05T08:07:06.919+09:30	Put event logs failed: InvalidSequenceTokenException
2021-03-05T08:07:06.995+09:30	rpc error: code = NotFound desc = error:pkg/identityserver/store:end_device_not_found (end device `******` not found): Error null
2021-03-05T08:07:06.996+09:30	END RequestId: 0da9902a-6c6e-45be-926d-545abef5c355
2021-03-05T08:09:10.175+09:30	START RequestId: 0da9902a-6c6e-45be-926d-545abef5c355 Version: $LATEST
2021-03-05T08:09:12.338+09:30	2021/03/04 22:39:12 Failed to get device: rpc error: code = NotFound desc = error:pkg/identityserver/store:end_device_not_found (end device `******` not found)
2021-03-05T08:09:12.338+09:30	rpc error: code = NotFound desc = error:pkg/identityserver/store:end_device_not_found (end device `******` not found): Error null
2021-03-05T08:09:12.339+09:30	END RequestId: 0da9902a-6c6e-45be-926d-545abef5c355
```

I then deleted the CloudFormation and tried to created a new one with the version 1.1.2 template, but it didn't succeed. The event logs show the following:
[image]

I though it was a problem with the API key so a generated a new one and tried to create the CloudFormation again, but the issue still persists.
Let me know if you need additional information or if you want me to run some tests.

Cheers

johanstokking, 2021-03-05:
The default Lambda execution timeout set by the AWS CDK is 3 seconds. That is not enough to create things in four components, especially considering latency between regions.

The timeouts are now adjusted in version 1.1.3. Please test with that CloudFormation stack.

---

> I though it was a problem with the API key so a generated a new one and tried to create the CloudFormation again, but the issue still persists.

This I can't reproduce. Make sure you select the **Community** CloudFormation template.

robsonos, 2021-03-08:
Hi @johanstokking ,

Please disregard the error with the CloudFormation creation. I just created a new one with 1.1.3 but the errors persist:

Create Thing Lambda function:
```
22021-03-08T08:53:34.496+09:30	START RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Version: $LATEST
22021-03-08T08:53:37.572+09:30	2021/03/07 23:23:37 Created on Identity Server
22021-03-08T08:53:38.283+09:30	2021/03/07 23:23:38 Failed to create on Join Server: rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask)
22021-03-08T08:53:38.536+09:30	2021/03/07 23:23:38 Rolled back create on Identity Server
22021-03-08T08:53:39.403+09:30	rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask): Error null
22021-03-08T08:53:39.404+09:30	END RequestId: a6077275-8908-4eec-94fb-0919cecef6fe
22021-03-08T08:53:39.404+09:30	REPORT RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Duration: 4908.04 ms Billed Duration: 4909 ms Memory Size: 128 MB Max Memory Used: 86 MB Init Duration: 358.20 ms
22021-03-08T08:54:44.582+09:30	START RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Version: $LATEST
22021-03-08T08:54:45.937+09:30	2021/03/07 23:24:45 Created on Identity Server
22021-03-08T08:54:46.628+09:30	2021/03/07 23:24:46 Failed to create on Join Server: rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask)
22021-03-08T08:54:46.896+09:30	2021/03/07 23:24:46 Rolled back create on Identity Server
22021-03-08T08:54:46.896+09:30	rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask): Error null
22021-03-08T08:54:46.896+09:30	END RequestId: a6077275-8908-4eec-94fb-0919cecef6fe
22021-03-08T08:54:46.896+09:30	REPORT RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Duration: 2310.92 ms Billed Duration: 2311 ms Memory Size: 128 MB Max Memory Used: 87 MB
22021-03-08T08:56:34.081+09:30	START RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Version: $LATEST
22021-03-08T08:56:35.536+09:30	2021/03/07 23:26:35 Created on Identity Server
22021-03-08T08:56:35.835+09:30	2021/03/07 23:26:35 Failed to create on Join Server: rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask)
22021-03-08T08:56:36.102+09:30	2021/03/07 23:26:36 Rolled back create on Identity Server
22021-03-08T08:56:36.103+09:30	rpc error: code = InvalidArgument desc = error:pkg/joinserver:field_mask (invalid field mask): Error null
22021-03-08T08:56:36.103+09:30	END RequestId: a6077275-8908-4eec-94fb-0919cecef6fe
22021-03-08T08:56:36.103+09:30	REPORT RequestId: a6077275-8908-4eec-94fb-0919cecef6fe Duration: 2018.19 ms Billed Duration: 2019 ms Memory Size: 128 MB Max Memory Used: 88 MB
```

Is this an error on eu1 and nam1 community as well?

Cheers

johanstokking, 2021-03-08:
@robsonos I cannot reproduce this.

Please attach a screenshot of the attributes as you configure them, before creating the thing.

See also https://www.thethingsindustries.com/docs/integrations/aws-iot/default/things/#creating-things

robsonos, 2021-03-08:
Hi @johanstokking 

I got some progress here. Apparently the device I was trying to create already exits somewhere within the servers ( I have been using the same devEUI, appEUI and appKey for the tests). That could possibly explain the logs from my previous comment. However, as this device is not listed anywhere in the console I cannot delete it and thus I cannot recreate it. 
I realised this when I tried to create this device on the console instead of from AWS IoT (using the same devEUI, appEUI and appKey) and got the following error:
[image]

After testing using another devEUI, Create Thing Lambda function executed without errors:
```
  | 2021-03-08T21:51:01.468+09:30 | START RequestId: 234344fb-a81b-4bf2-8686-2d716e263c6e Version: $LATEST
  | 2021-03-08T21:51:04.524+09:30 | 2021/03/08 12:21:04 Created on Identity Server
  | 2021-03-08T21:51:05.217+09:30 | 2021/03/08 12:21:05 Created on Join Server
  | 2021-03-08T21:51:05.534+09:30 | 2021/03/08 12:21:05 Created on Network Server
  | 2021-03-08T21:51:06.316+09:30 | 2021/03/08 12:21:06 Created on Application Server
  | 2021-03-08T21:51:06.316+09:30 | 2021/03/08 12:21:06 Created end device
  | 2021-03-08T21:51:06.352+09:30 | END RequestId: 234344fb-a81b-4bf2-8686-2d716e263c6e
```

For reference, I am using the following attributes in AWS IoT:
[image]

I believe the last change you applied fixed the issue, but I still need to reuse that devEUI. Is there a way to delete this device other than using the console?

Cheers.

johanstokking, 2021-03-08:
OK, that all looks good.

> I believe the last change you applied fixed the issue, but I still need to reuse that devEUI. Is there a way to delete this device other than using the console?

Hmm, that's a tricky one as we currently don't automatically purge orphan devices (that is, devices that are deleted from the Identity Server but that are still in Application Server) for example. References #2735 

I'll contact you on TTN Slack to resolve this.

closed 2021-03-08 by johanstokking

robsonos, 2021-03-08:
Thanks, Johan.

I keep an eye on that thread. I did notice that I couldn't reuse the same application ID after deleting it. It is good to know that there is a thread about that too.

Cheers