TheThingsNetwork/lorawan-stack #3890: Move end devices between applications

kschiffer, 2021-03-04:
#### Summary
We currently don't allow moving end devices between different applications in the Console. This functionality should be added.

#### Why do we need this?
There is a use case where end devices are first registered on a testing/QA application and after approval they can be transferred to a "production" application. Right now, it's necessary to manually delete the device and re-enter the device registration data in the next application.

#### What is already there? What do you see now?
- Device import

#### What is missing? What do you want to see?
- A simple way to move end devices from one application to another.

#### How do you propose to implement this?
- An entry inside the end device general settings enabling to transfer a device to another application
- Possibility to select multiple devices (via a checkbox) in the applications device list, to similarly mark them for transfer into another application

#### How do you propose to test this?
End-to-end tests

#### Can you do this yourself and submit a Pull Request?
The exact flow needs to be fleshed out in a wireframe by me. I can then decide who can work on the implementation.

johanstokking, 2021-03-16:
This is what claiming is for, and that already works perfectly per device.

What I think we need instead is:

- Single end device claiming support in the Console (https://github.com/TheThingsNetwork/lorawan-stack/issues/3770)
- Assign claim authentication codes in bulk
- Export end devices with claim authentication code in bulk
- Bulk claiming

KrishnaIyer, 2021-03-16:
I can pick up the `dcs` part. There have been similar requests for gateways as well.

johanstokking, 2021-03-16:
I think I need to mark this as a duplicate of https://github.com/TheThingsIndustries/lorawan-stack/issues/2345 as we will (only) transfer of devices via claiming.

kschiffer, 2021-03-16:
Is this really the way to go for moving devices between two applications of the same owner? Without doing a lot of research on the technicalities, it seems like quite an overhead in terms of UX, to be able to transfer devices within the same ownership domain.

johanstokking, 2021-03-16:
> Is this really the way to go for moving devices between two applications of the same owner?

I don't think it should matter whether it is the same owner or not. I agree that transferring devices from one app to another is a simple use case, but let's not build a specific process for that if device claiming can do the same and much more.

Technically, devices are being deleted from IS, JS, NS and AS, and then created again. That is a vulnerable process that needs lots of security checking, validation and rollback. This is already built with device claiming and I think that we should leverage that infrastructure.

I do agree that working with claim authentication codes and application authorization for claiming is cumbersome, if we know that the caller has access to both the source and the destination application. In that case, I would prefer adjusting things in DCS so that it also works without claim authentication codes and application authorization for claiming.

kschiffer, 2021-09-23:
Reopening this since with [progressive insight](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/187) and the meeting held today we determine that there should be a way to move end devices between applications without using the claiming flow.

johanstokking, 2021-09-24:
@NicolasMrad can you write a quick summary of what we discussed offline?

NicolasMrad, 2021-09-24:
After discussing the use cases internally and with CyberEye, and after discussing the technical feasibility of the solution:

**Use Cases**
1. Migration of devices from TTS Community Edition to TTS Cloud tenant with active sessions
2. Migration of devices from one tenant to the other with active sessions (in the same network)
3. Migration of devices from one application to the other with/without active sessions (in the same network+tenant+cluster)
4. Migration of devices from one cluster of their tenant (say eu1) to the other cluster (nam1/au1) with active sessions (in the same network+tenant)

Basically if a customer wants to move from TTS Community edition to TTS Cloud, then they might want to move all their devices in most cases. And if they want to move some devices from one tenant to other tenant (in same network) based on their business model, they might want to move batches of devices to a different customer or tenant each time.

**Technical Feasibility**

1. Migration of devices from one application to the other with active sessions within the same tenant and cluster can be done.
2. Moving of sessions is not possible for the following:
                 - From one tenant to the other within the same cluster
                 - Within the same tenant but from one cluster to the other
                 - From one tenant to the other and from one cluster to the other (such as from community to TTSC)
3. Point (2) is due to a design issue by LoRaWAN and not a limitation from our side
4. For the cases in point (2), we will be using device claiming which will not maintain the session and create a new one #3770
5. Implementation will be affected by the database restructuring https://github.com/TheThingsIndustries/lorawan-stack/issues/2256
6. Discussion on wether to have the option of selecting manually the devices to be moved, or just have either the option to move one device at a time or all the devices in the application.

kschiffer, 2021-12-20:
Closing this as duplicate of TheThingsIndustries/lorawan-stack#2916

closed 2021-12-20 by kschiffer