TheThingsNetwork/lorawan-stack #3891: Add configurable expiry to authentication/session cookies

kschiffer, 2021-03-04:
#### Summary
We currently don't use any expiry for our session cookie in the IS, as well as for the auth cookie of the oauth clients. This should be fixed and the exact value should be configurable.

#### Why do we need this?
It's not always desirable to have session cookies without expiry. (See https://github.com/TheThingsIndustries/lorawan-stack-support/issues/328)

#### What is already there? What do you see now?
Session and auth cookies using now expiry, meaning that they will stay valid until the end of the browser session.

#### What is missing? What do you want to see?
1.  Expiring the Account App session cookie (we currently don't have an expiry set, which means that the session expires when the browser session ends)
2.  Expiring the auth cookie of the console OAuth client, so that the client cannot use the refresh token anymore.
3.  Use `ExpiresAt` prop for the user session during creation (currently not set)
4.  Making these expiries configurable (I'd say we can use the same config for all of these)

#### How do you propose to implement this?
This is a matter of extending the IS config and using this config in the creation of the cookies and tokens above.

#### How do you propose to test this?
Unit tests as well as some special end-to-end tests, though we'd likely need to run the stack with custom configuration to test this properly.

#### Can you do this yourself and submit a Pull Request?
Yes.

htdvisser, 2021-03-09:
See also https://github.com/TheThingsIndustries/lorawan-stack/issues/1411

johanstokking, 2021-07-20:
We see that this can be relevant for some specific scenarios, but we need thumbs up to schedule this to a version milestone.

kschiffer, 2022-03-21:
Closing this in favor of TheThingsIndustries/lorawan-stack#1411

closed 2022-03-21 by kschiffer