TheThingsNetwork/lorawan-stack #3894: Disable UDP remote address check by default

johanstokking, 2021-03-05:
#### Summary

Disable the UDP remote address check by default.

#### Why do we need this?

Because the number of gateways that are changing remote addresses seems to be increasing rapidly, sometimes even for the upstream and downstream session, which leads to an unusable gateway.

#### What is already there? What do you see now?

The `gs.udp.addr-change-block` is 1 minute

#### What is missing? What do you want to see?

Make it `0` to disable it.

#### How do you propose to implement this?

1. Change default configuration: https://github.com/TheThingsNetwork/lorawan-stack/blob/51606a79fd661c2740e13ef06a1bcaa07c1b064a/pkg/gatewayserver/io/udp/config.go#L57. When 0, the address check isn't enabled at all (https://github.com/TheThingsNetwork/lorawan-stack/blob/51606a79fd661c2740e13ef06a1bcaa07c1b064a/pkg/gatewayserver/io/udp/udp.go#L58). BTW, while you're at it, remove that `== true` thing that got my eyes bleeding
2. Add deprecation notice in changelog
3. Check that our deployment scripts do not set this setting so that it will get configured anyway

#### How do you propose to test this?

This is already tested in some production environments

#### Can you do this yourself and submit a Pull Request?

Can review

KrishnaIyer, 2021-03-08:
Refs; https://github.com/TheThingsIndustries/lorawan-stack-aws/pull/376

johanstokking, 2021-03-08:
> Refs; [TheThingsIndustries/lorawan-stack-aws#376](https://github.com/TheThingsIndustries/lorawan-stack-aws/pull/376)

OK good for ops, but we need a default in the codebase too.

KrishnaIyer, 2021-03-08:
> OK good for ops, but we need a default in the codebase too.

Yes
https://github.com/TheThingsNetwork/lorawan-stack/pull/3906

PR #3906 (2021-03-08): Disable UDP IP firewall by default

#### Summary

Disable UDP IP firewall by default. Closes #3894 

#### Changes

- Set the default value of 0 (0 is equivalent to `0 * time.Second` for `time.Duration`)
- Add note on security of new setting.

#### Testing

No new functions added. Only config value. 

##### Regressions

I don't think there'd be any.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments  [... cut]
closed 2021-03-08 by KrishnaIyer