TheThingsNetwork/lorawan-stack #3909: Possibility to configure NetIDs in the stack

i3mSuperMan, 2021-03-09:
#### Summary
Basic Station/LNS protocol defines a `router_config` message, and it contains a field called `NetID`. This field allows filtering the device addresses the gateway will forward to the stack. Hence, we are seeking support on the possibility to configure these `NetIDs` in the stack, which will be forwarded to the Basic Station gateways in the `router_config` message

A stack-wide setting with the possibility to override the list for every gateway would be nice.
#### Why do we need this?
To filter the device addresses that the Gateway will forward to the stack.
#### What is already there? What do you see now?
In LNS, right after the Web Socket connection has been established, The LNS shall respond to a `version message` with a `router_config` message to specify a channel plan for the Station and define some basic operation modes.
```
{
  "msgtype" : "router_config"
  "NetID" : [ INT, .. ]
  "JoinEui" : [ [INT,INT], .. ] // ranges: beg,end inclusive
  "region" : STRING // e.g. "EU863", "US902", ..
  "hwspec" : STRING
  "freq_range" : [ INT, INT ] // min, max (hz)
  "DRs" : [ [INT,INT,INT], .. ] // sf,bw,dnonly
  "sx1301_conf": [ SX1301CONF, .. ]
  "nocca" : BOOL
  "nodc" : BOOL
  "nodwell" : BOOL
}
```
The fields `NetID` and `JoinEui` are used to filter LoRa frames received by the Station. NetID is a list of NetID values that are accepted. Any LoRa data frame carrying a NetID other than those listed will be dropped.
Ref: https://doc.sm.tc/station/tcproto.html#router-config-message
#### What is missing? What do you want to see?
Possibility to configure the `NetID` field of the `router_conf` message in the Lora Basic station gateways at the stack level.
#### Environment
TTS v3.11.1 Self Hosted
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
Nope. @KrishnaIyer

onsolutionjames, 2022-03-31:
Just wondering if there are any updates on this? Is this feature likely to be implemented in TTS at all?

racosa, 2023-12-06:
This would be important for us too. Are there any updates?

I've just asked a similar question on the Multitech forum:

https://www.multitech.net/developer/forums/topic/is-it-possible-to-filter-packets-when-using-basics-station/

KrishnaIyer, 2025-04-30:
This is tracked here: https://github.com/TheThingsIndustries/lorawan-stack/issues/2703

closed 2025-04-30 by KrishnaIyer

StevenCellist, 2025-04-30:
@KrishnaIyer that repository is a closed repository that we cannot see.