TheThingsNetwork/lorawan-stack #3913: Disable uplink simulation and downlink message sending when skipping payload crypto

johanstokking, 2021-03-10:
#### Summary

Disable uplink simulation and downlink message sending when skipping payload crypto

#### Why do we need this?

When simulating uplink, the simulated payload would still be encrypted. I don't think it's of great added value to support uplink simulation with encrypted FRMPayload.

When sending downlink, AS doesn't have a FCntDown so it always fails. Also, the same problem applies: it would already be encrypted.

#### What is already there? What do you see now?

The features are enabled even when end-to-end encryption (skip payload crypto) is enabled on either end device or application level.

#### What is missing? What do you want to see?

Let's begin with returning errors from the API. We might add nice Console support too, but that's a bit goldplating as this is pretty much and edge case.

#### How do you propose to implement this?

- For uplink simulation, when fetching device identifiers, also fetch `skip_payload_crypto` for the end device and the application. When enabled, return an error
- When sending downlink messages, this might be a little trickier, because we use the normal API that integrations also use to push/replace downlink. I'm not sure what's the best way to implement there: that might need to go via Console

#### How do you propose to test this?

Locally

#### Can you do this yourself and submit a Pull Request?

Can review and test

adriansmares, 2021-04-09:
Simulated uplinks go via `as.handleSimulatedUp`

adriansmares, 2021-04-09:
I've added the simulated uplink error in https://github.com/TheThingsNetwork/lorawan-stack/pull/4045 .

With that being said, I personally believe we should just hide the `Messaging` tab in the Console if `link.skip_payload_crypto` or `device.skip_payload_crypto_override.value` are true. cc @kschiffer

kschiffer, 2021-04-20:
Indeed the messaging tab should be disabled in a way that the tab is still there, but the form is disabled with a notification reading that message simulation is disabled for devices that skip payload crypto.

closed 2021-05-11 by asmulko