TheThingsNetwork/lorawan-stack #3921: Device rejoin and can not update CFlist

viethoa14, 2021-03-12:
#### Summary

Device disconnect to NS about several day, device maybe rejoin (still devaddr, netid). After device connect again NS, device only work at boot frequencies. NS send ADR request, device ack with datarate, tx power index, reject channel mask. NS continue send ADR request, no send CFlist. But with other provider NS, it continue send CFlist.  

#### Steps to Reproduce

1. NS can send to request device join again or not? 
2. NS can ignore reject channel mask as other provider NS, and continue send CFlist 

#### What do you see now?

Log on other provider

[image]

[image]

[image]

log on Things Stack

[image]

#### What do you want to see instead?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

viethoa14, 2021-03-17:
thanks for your update milestones. Can you tell me the current status this issue?

1.  You have not yet enough time for test now?
2.  You have found problem but have not enough time for fix?
3.  You have test but don't seen problem...?

viethoa14, 2021-05-25:
dear @johanstokking 
I try to update cflist by command on two server, but it is not ok: 

ttn-lw-cli end-devices set app1  test-412  --pending-mac-state.pending-join-request.cf-list.freq 921800000
error:pkg/networkserver:field_value (invalid value of field `pending_mac_state.lorawan_version`)
    field=pending_mac_state.lorawan_version
    correlation_id=76efa11709874ccba8a51ce8816fb906

ttn-lw-cli end-devices set app1 c4cd8297d736b904 --pending-mac-state.pending-join-request.cf-list.freq 921800000
error:pkg/rpcmiddleware/validator:field_mask_paths (forbidden path(s) in field mask)
    forbidden_paths=[pending_mac_state.pending_join_request.cf_list.freq]
    correlation_id=fe9a3f2c3a134050ae43e27846fdb563

I can do update cf_list frequency by command whether or not? if it is ok, please show me detail that command or send message PAYLOAD downlink! 

Thanks!

johanstokking, 2021-05-25:
> 5\. When to limit uplink, device rejoin, device only save some information: dev-eui, netid, networkskey; it lost some information CFlist that updated by NS...and only action on default frequencies. I resend some uplink before turn on gateway

Can you clarify what you mean with this exactly?

The end devices joins again, or does it not join again? If it joins again, why does it save the NetID and NwkSKey?

---

Do you know why the end devices rejects the `NewChannelReq` commands?

viethoa14, 2021-05-25:
When device can not connect to NS, after some uplink or limit time we seen device maybe restart but it still save Dev addr, netid (i see on its log debug)
When device can connect to NS, it still works properly with NS, but only work with default frequency, and at SF12. 
When device send first uplink after it connected to NS again, NS send to ADR requested, device only ack with Tx index, datarate but reject channel mask. NS send again...
On other provider'NS that we are using, after device answer ADR request, NS send Newchannel_request.

johanstokking, 2021-05-25:
> When device can not connect to NS, after some uplink or limit time we seen device maybe restart but it still save Dev addr, netid (i see on its log debug)
> When device can connect to NS, it still works properly with NS, but only work with default frequency, and at SF12.

This seems to be normal behavior, driven by the ADR algorithm in the end device. It increases the TX power, decreases the data rate, and finally reverts to factory default channels.

If the device were joining, the MAC state would be reset and the DevAddr would change. So the device isn't joining.

> datarate but reject channel mask

Do you know why?

> NS send again...we did not seen Newchannel_request
On other provider'NS that we are using, after device answer ADR request, NS send Newchannel_request. nb_trans=1

Can you clarify this? You mean that The Things Stack does _not_ send `NewChannelReq`, while other NS providers are sending it? Which providers are that?

What is the frequency plan that you configured for these end devices exactly? Which regional parameters version are you uing?

viethoa14, 2021-05-26:
device reject channel mask maybe device only work on default frequency, but NS request "ff" for channel mask. 
And NS continue to send ADR request again, device continue reject..again... (with other provider we seen, NS send only one "ADR request" in this situation, after then it send request update cf_list)
We are using AS923, and version 1.0.2 for devices.

johanstokking, 2021-05-26:
@viethoa14 I need to know the frequency plan, not just the band. Also the regional parameters version. Please retrieve this information in the Console under Network Layer settings.

viethoa14, 2021-05-26:
Sorry for lack of information, 
[image]

johanstokking, 2021-05-26:
Thanks.

Are you getting a `CFList` in the join-accept? If so, what does it look like?

The frequency plan you're using is https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/AS_920_923.yml. Are you sure this end device can use these frequencies?

The `ChMask` of `0x00FF` means that it enables all first eight channels. That is OK.

I don't know why the end device rejects that instruction by setting `ChannelMaskAck` to `0`. Because of that, NS sends another instruction with `ChMask` of `0x0003` (enable only channels 1 and 2, the default join channels) and then tries to individually create the other six channels through `NewChannelReq`.

---

What is the behavior you see on other LNSs? Which LNSs are that?

viethoa14, 2021-05-27:
I'sure, my device works with that frequency plan, we have tested (view log on both device and NS) for along time. 
log boot on TTN console
[image]

[image]

I can show log debug on my device for this issue:
* on Things Stack network: 
- device boot successfully, this packet contain cflist, netid... 
rx receive on device when it is join-accepted (hex data): 

snr=8, RssiValue=-92
rxdatalen=33,rxdata= 20 62 8e e3 82 18 70 16 c6 0f 8e b0 34 0a 76 fb 46 ff 89 68 0c da 1d 36 67 bf 7e cc 2e 39 44 10 4d
LoRaMacNetID = C00014,LoRaMacDevAddr=FC0050F1,Rx1DrOffset=0,Rx2Channel.Datarate=2,ReceiveDelay1=5000
applyCFList.Size=16
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0
 
-  When device work with default frequency (after it can not connect to NS for along time)
rx receive on device after it send first uplink when it connected NS again  (ADR request enqueue)

tmr-20002b34
snr=7, RssiValue=-94
rxdatalen=17,rxdata= 60 f1 50 00 fc 85 07 00 03 43 fc 03 01 2e 75 3c 9a
DownlinkDwellTime = 0
address = fc0050f1
OnRadioRxDone:MType = 3 ,fcnt=7,DownLinkCounter=5
ADR_REQ ChMask = 3fc
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

* On other providers:

-  When device work with default frequency (after it can not connect to NS for along time)
rx receive on device after it send first uplink when it connected NS again (log on web management as attachment previous)

+ downlink 1 (ADR request equeue) , this device reject same as TTN network: 

snr=8, RssiValue=-102
rxdatalen=18,rxdata= 60 3e 51 00 fc 00 04 00 00 f2 05 40 63 2c 2c 4a 42 25
DownlinkDwellTime = 0
address = fc00513e
OnRadioRxDone:MType = 3 ,fcnt=4,DownLinkCounter=3
payload mac commands :: port= 0,frameLen =5,data =0x3 0x0 0xFF 0x0 0x2
ADR_REQ ChMask = ff
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0 

+ downlink 2 (update cflist, but ADR_REQ_chaMask = 3), device accept 

snr=8, RssiValue=-93
rxdatalen=36,rxdata= 60 3e 51 00 fc 00 05 00 00 ca 78 33 21 b9 62 49 2d b8 9c e3                                                                                  35 b3 d1 16 62 3b 94 c4 5f 2d 69 96 8a 8c b5 32
DownlinkDwellTime = 0
address = fc00513e
OnRadioRxDone:MType = 3 ,fcnt=5,DownLinkCounter=4
payload mac commands :: port= 0,frameLen =23,data =0x3 0x0 0x3 0x0 0x2 0x7 0x2 0                                                                                        xD0 0xA7 0x8C 0x50 0x7 0x3 0xA0 0xAF 0x8C 0x50 0x7 0x4 0x70 0xB7 0x8C 0x50
ADR_REQ ChMask = 3
ADR_REQ ChMask = 3
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

+ downlink 3 (update continue clist remain)

Event 1 20002bd4
snr=7, RssiValue=-104
rxdatalen=31,rxdata= 60 3e 51 00 fc 00 06 00 00 de 7d ce f7 9e 59 1b 99 e7 8a 52                                                                                         29 d2 f1 c2 73 ac 89 d1 70 b1 88
DownlinkDwellTime = 0
address = fc00513e
OnRadioRxDone:MType = 3 ,fcnt=6,DownLinkCounter=5
payload mac commands :: port= 0,frameLen =18,data =0x7 0x5 0x40 0xBF 0x8C 0x50 0                                                                                        x7 0x6 0x10 0xC7 0x8C 0x50 0x7 0x7 0xE0 0xCE 0x8C 0x50
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

johanstokking, 2021-05-27:
Thanks.

> 1. on Things Stack network:
> 
> * device boot successfully, this packet contain cflist, netid...
>   rx receive on device when it is join-accepted (hex data):
> 
> snr=8, RssiValue=-92
> rxdatalen=33,rxdata= 20 62 8e e3 82 18 70 16 c6 0f 8e b0 34 0a 76 fb 46 ff 89 68 0c da 1d 36 67 bf 7e cc 2e 39 44 10 4d
> LoRaMacNetID = C00014,LoRaMacDevAddr=FC0050F1,Rx1DrOffset=0,Rx2Channel.Datarate=2,ReceiveDelay1=5000
> applyCFList.Size=16
> LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

What are the contents of CFList? The join-accept in `rxdata` is encrypted so can't see that here.

> * When device work with default frequency (after it can not connect to NS for along time)
>   rx receive on device after it send first uplink when it connected NS again  (ADR request enqueue, device reject this downlink)
> 
> tmr-20002b34
> snr=7, RssiValue=-94
> rxdatalen=17,rxdata= 60 f1 50 00 fc 85 07 00 03 43 fc 03 01 2e 75 3c 9a
> DownlinkDwellTime = 0
> address = fc0050f1
> OnRadioRxDone:MType = 3 ,fcnt=7,DownLinkCounter=5
> ADR_REQ ChMask = 3fc
> LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

The problem here is the `ChMask` of `0x03FC`, those are channels `0000001111111100`, i.e. 8 channels above the 2 default join channels. That assumes that there are 10 channels in the end device, which is probably not the case. That may be the reason why the end device rejects the channel mask in `LinkADRAns`.

Can you please provide a full transcript of PHYPayload including session keys?

Also, can you provide the output of `ttn-lw-cli end-devices get <app-id> <device-id> --mac-state` _before_ and _after_ the end device sends the uplink after a long time?

viethoa14, 2021-05-27:
dear
- CFlist contain: netid, rx2_dr, cflist (5 channels)... my mean my device boot successfully 
- With chMask, my device reject ADR request enqueue both provider (Things Stack and That we is using for service). Please see section other provider, in downlink 1. my NS also reject, though ADR_REQ ChMask = ff. On Things Stack NS, because default frequency is not same with my device, so ChMask is 0x03FC.
- "Can you please provide a full transcript of PHYPayload including session keys?" Can you explain your mean? please check for attach file, if it is not ok, could you support for me
[Full_PAYLOAD.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6552890/Full_PAYLOAD.txt)
[log_mac_state.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6552891/log_mac_state.txt)

johanstokking, 2021-05-27:
Yeah so this is a bit weird, indeed I see 10 channels in de `desired_parameters.channels`.

How did you create the device exactly?

Can you reproduce this step by step with a (new) device, or by recreating this device?

viethoa14, 2021-05-27:
Dear 
I think it is not issue. I am changing configuration to channel=8. I still think, problem is ADR request enqueue. I see both TTN and my provider system, they reject ADR request, after than my provider continue send cflist.
I will send log when i finish changing paramater

Thanks for your support

viethoa14, 2021-05-28:
we have recreated device, chmask=8, but that issue still exists. Please see log file.

- rx receive on device at boot time:

tmr-20002b20
====rx1 window=: rxdr=2
freqoffset=0,physical=921400000
snr=7, RssiValue=-88
rxdatalen=33,rxdata= 20 19 a8 22 4d c1 7f 83 cd 0d 82 36 92 c9 5b 71 3a d8 a9 06                                                                                         ba d4 66 fa d2 b6 df 88 d7 c7 0a ed b5
LoRaMacNetID = C00014,LoRaMacDevAddr=FC005350,Rx1DrOffset=0,Rx2Channel.Datarate=                                                                                        2,ReceiveDelay1=5000
applyCFList.Size=16
LoRaMacState=1 IsRunning=0 McpsIndSkip=0 AckReceived=0 RxData=0

- rx receive on device at after along time (device reject this packet)

tmr-20002b34
snr=9, RssiValue=-98
rxdatalen=17,rxdata= 60 50 53 00 fc 85 0a 00 03 52 ff 00 01 62 39 af 1e
DownlinkDwellTime = 0
address = fc005350
OnRadioRxDone:MType = 3 ,fcnt=10,DownLinkCounter=9
ADR_REQ ChMask = ff

[Payload_session_key.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6558740/Payload_session_key.txt)
[mac-state-before-disconnect.log](https://github.com/TheThingsNetwork/lorawan-stack/files/6558743/mac-state-before-disconnect.log)
[mac-state-after-along-time.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6558823/mac-state-after-along-time.txt)

johanstokking, 2021-05-28:
> I think it is not issue. I am changing configuration to channel=8.

We need more details here. What do you mean with "changing configuration to channel=8"?

>  I still think, problem is ADR request enqueue.
> [...]
> 
> ADR_REQ ChMask = ff

This is in fact the correct channel mask.

So, both The Things Stack and "the other provider" send the same `LinkADRReq`. The end device rejects that. Then TTS behaves differently on the rejection than your provider, right?

From what I understand, The Things Stack is trying to create the individual channels with `NewChannelReq` after. That is by design.

---

The data downlink message message, with PHYPayload `60505300fc850a000352ff00016239af1e` has FOpts `0352FF0001`, meaning:

- `LinkADRReq`
   - `TXPower`: `2`
   - `DataRate`: `5`
   - `ChMask`: channels 0 to 7 enabled
   - `NbTrans`: `1`
   - `ChMaskCntl`: `0` (meaning that `ChMask` applies to uplink channels)
 
Why is the device rejecting this?

What device is this? Is this certified for AS923? Is it certified for a country to operate in AS920-923 frequencies?

---

What I still don't understand, is how The Things Stack has channels 2 to 9 enabled in the desired parameters, in the MAC state you copied in earlier.

---

To continue support here, I need the full reproduction steps. That is either a screencast of how the device is created or the CLI commands, as well as the event stream (`ttn-lw-cli events subscribe --application-id <app-id>`) and the MAC state before and after the disconnection.

viethoa14, 2021-05-28:
> We need more details here. What do you mean with "changing configuration to channel=8"?

I add more channel factory on web console

>So, both The Things Stack and "the other provider" send the same `LinkADRReq`. The end device rejects that. Then TTS behaves differently on the rejection than your provider, right?

I Think so, My provider NS only one send ADR request enqueue, though it is reject but NS continue to send cflist request. 

> From what I understand, The Things Stack is trying to create the individual channels with `NewChannelReq` after. That is by design.

I do not see on log. NS send ADR request enqueue  and don't stop

> Why is the device rejecting this?

Maybe now it work on default frequency, i will check again

> What device is this? Is this certified for AS923? Is it certified for a country to operate in AS920-923 frequencies?

I will check it later

> To continue support here, I need the full reproduction steps. That is either a screencast of how the device is created or the CLI commands, as well as the event stream (`ttn-lw-cli events subscribe --application-id <app-id>`) and the MAC state before and after the disconnection.

I am doing it and send to you later

viethoa14, 2021-05-28:
picture for create NE
[image]
[image]
[image]

viethoa14, 2021-06-08:
I try sending mac command to device for update channel list (ex: 03531F0001070540bf8c50070610c78c500707e0ce8c50), i try finding from ttn-lw-cli, but i don't see any command suitably, on web console only use for application specific (fport = 1 to 223). Can you show me, how do i update manually cflist?

johanstokking, 2021-06-10:
@viethoa14 thanks for providing the information. The key here is that you enter the factory preset frequencies manually. You should not do that. These frequencies are implied by the AS923 band, and additional frequencies come through the frequency plan: https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/f77e7cc095ca01d173bdae0b2f5d2277e69c7d53/AS_920_923.yml

I see you also contribute to a Viet Nam channel plan (https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/30). Are you trying to manually change the frequencies? Is the end device rejecting the frequencies from the channel plan maybe?

viethoa14, 2021-06-10:
Thanks for your support, we have tested TTN system about four month, and now we have some problems, so we can not move it to production. TTN has many advantages, we hope that problems will been fixed soon.

johanstokking, 2021-06-10:
OK. So let's track this through https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/30 and #3150.

closed 2021-06-10 by johanstokking