TheThingsNetwork/lorawan-stack #3923: Application overview webpage constantly refreshes and flashes

kurtmc, 2021-03-15:
#### Summary

When I navigate to the application overview pager for our lorawan-stack deployment, the page is refreshing constantly and flashing on the screen, it happens at a really high frequency.

#### Steps to Reproduce

1. Run v3.11.2
2. navigate to application overview page

#### What do you see now?

[image]

#### What do you want to see instead?

Stable webpage.

#### Environment

v3.11.2

#### How do you propose to implement this?

Not sure.

#### How do you propose to test this?

Deploy new versions to my test environment.

#### Can you do this yourself and submit a Pull Request?

No.

johanstokking, 2021-03-16:
That doesn't look good indeed.

Do you see errors or log messages in the console of the browser (via Developer Tools)?

Which browser are you using exactly?

kurtmc, 2021-03-16:
Just tested this again on Firefox and Google Chrome, both have the exact same issue. Errors in the javascript console, firefox:
```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://localhost:1885/api/v3/edtc/formats. (Reason: CORS request did not succeed).
```
chrome:
```
Failed to load resource: net::ERR_CONNECTION_REFUSED localhost:1885/api/v3/edtc/formats:1
GET http://localhost:1885/api/v3/edtc/formats net::ERR_CONNECTION_REFUSED
```

I possibly have something configured incorrectly, maybe I have set the hostname of something to `localhost` instead of it's real hostname.

kurtmc, 2021-03-16:
I guess I need to set `TTN_LW_CONSOLE_UI_EDTC_BASE_URL`

bafonins, 2021-03-16:
@kurtmc This seems to be a configuration issue. Please try setting the `TTN_LW_CONSOLE_UI_EDTC_BASE_URL` variable.

However, this indeed the webui issue. The Console should handle such cases and render the error view instead of getting stuck in a refresh loop.

kurtmc, 2021-03-16:
I have now set `TTN_LW_CONSOLE_UI_EDTC_BASE_URL` and it resolves the issue I was having. In my config I currently have a bunch of base urls set, looking something like this:
```
            - TTN_LW_CONSOLE_UI_AS_BASE_URL=https://${host}/api/v3
            - TTN_LW_CONSOLE_UI_GS_BASE_URL=https://${host}/api/v3
            - TTN_LW_CONSOLE_UI_IS_BASE_URL=https://${host}/api/v3
            - TTN_LW_CONSOLE_UI_JS_BASE_URL=https://${host}/api/v3
            - TTN_LW_CONSOLE_UI_NS_BASE_URL=https://${host}/api/v3
            - TTN_LW_CONSOLE_UI_EDTC_BASE_URL=https://${host}/api/v3
```

Am I doing something wrong? Can this be set with a single flag instead of 6? This would prevent this kind of issue in the future if a new service is added in a new lorawan version and I don't know about it.

bafonins, 2021-03-16:
As far as I can see this should be set by default https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.11/cmd/internal/shared/console/config.go#L47-L56.

cc @htdvisser

kurtmc, 2021-03-16:
@bafonins It looks like that is hardcoded to localhost: https://github.com/TheThingsNetwork/lorawan-stack/blob/32ad18b2bac1359d9c8e7669d1466362a7ade154/cmd/internal/shared/config.go#L137-L140

Also, looking at your link, should I also be setting base URLs for `QRG` and `GCS`?

closed 2021-03-23 by bafonins