TheThingsNetwork/lorawan-stack #3931: Gateway opt out for Packet Broker peering

johanstokking, 2021-03-16:
#### Summary

Opt out specific gateways for peering through Packet Broker

#### Why do we need this?

Some users in The Things Network might not want to open their gateways for peering with commercial networks.

Also, some commercial networks may not want some gateways to peer with other networks.

#### What is already there? What do you see now?

With #3771, routing policies can be configured for the entire network or tenant.

#### What is missing? What do you want to see?

A setting per gateway to opt out of peering.

#### How do you propose to implement this?

- Add a new field to `Gateway`: `bool disable_packetbroker_forwarding` or something. I don't think it should be a `BoolValue`; it would really suppress the GS forward configuration, and not override it if GS doesn't have GS forwarding to PB configured
- On gateway connect, let GS fetch the field. GS already fetches a bunch of fields so that's pretty easy. https://github.com/TheThingsNetwork/lorawan-stack/blob/27720c5608ab7897c5dcb349ba5559ee86f48f8f/pkg/gatewayserver/gatewayserver.go#L462-L470
- We probably want to poll the Entity Registry for changes. Not only for this value, but also for the other fields, and let the IO middleware account for config changes.
- We would need to wire in a check for this value in the PB upstream handler, or better yet, before it gets sent to the upstream handler. We can also consider disabling the PB upstream handler for the connection, but if we want to make this dynamic, we need to be able to start and stop it, which is quite tricky

To discuss:

Should we combine the debouncing of updating gateway location, with some periodic rpc from GS to ER that updates the gateway location and gets gateway settings, in one go? cc @htdvisser @neoaggelos 

#### How do you propose to test this?

Local testing

#### Can you do this yourself and submit a Pull Request?

@neoaggelos to implement

neoaggelos, 2021-03-16:
One of the main issues with polling the Entity Registry is that it does not play well with unauthenticated gateways (UDP). 

References #2821

htdvisser, 2021-03-16:
Getting stuff from IS with cluster auth is already possible. On the GS side, this issue simply adds an extra field to that "getting stuff", and handling that field. 

Anything else is out of scope.

- The "polling for updates from ER" discussion is, as @neoaggelos already commented, in https://github.com/TheThingsNetwork/lorawan-stack/issues/2821
- The "writing stuff to ER from unauthenticated gateways" is going to get easier when we can prevent unauthenticated connections for specific gateways (https://github.com/TheThingsNetwork/lorawan-stack/issues/1436). If we have that, we can allow ClusterAuth to write stuff for gateways that do allow unauthenticated connections.

johanstokking, 2021-03-17:
OK, I scheduled #2821 also for Q2. Let's scope this issue only to:

- Adding `disable_packetbroker_forwarding` to `Gateway`
- Getting the field in GS, just like other settings are retrieved (currently only on connect, with #2821 also during a connection)
- Skip the Packet Broker upstream handler in `handleUpstream()`. Still consider not starting the PB upstream when forwarding is disabled for the connection, but make sure we don't bloat the codebase with dynamically starting and stopping upstream handlers if we only win one idle goroutine per connected gateway with PB forwarding disabled (i.e. the memory cost is very minimal)

PR #4396 (2021-07-20): Gateway opt out of Packet Broker forwarding

#### Summary

Closes #3931 

#### Changes

- Add `DisablePacketBrokerForwading` field to gateways.
- Skip `packetbroker` upstream host if needed.
- Console support

[image]

#### Testing

:monkey: 

##### Regressions

There should not be any.

#### Notes for Reviewers

- Default behaviour is backwards compatible, gateways forward to Packet Broker by default.
- Gateway needs to reconnect after updating config.
- 
#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unre
closed 2021-07-20 by neoaggelos