TheThingsNetwork/lorawan-stack #3939: Incorrect maximum payload sizes

jamesl-dm, 2021-03-17:
#### Summary
The maximum payload sizes the V3 stack will send appear to be 5 bytes too small.

#### Steps to Reproduce
Try to send a maximal size user payload at SF7.

#### What do you see now?
The largest payload I can get the network to deliver is 217 bytes.

#### What do you want to see instead?
The correct maximum size is 222 bytes for repeater compatibility.

#### Environment
AU915 subband 2, but should apply to any

#### How do you propose to implement this?
I think the problem is a failure to convert between MACPayload sizes and PHYPayload sizes at:
https://github.com/TheThingsNetwork/lorawan-stack/blob/9100f50012bc8acd27e18286e109cda690d11d98/pkg/networkserver/downlink.go#L1182
But the code is a bit too complex for me to follow, so it may be somewhere else, or in more than one place.
Someone who understands the internals will need to look at it in detail.

#### How do you propose to test this?
When fixed, the network should deliver a 222 byte downlink at SF7/500, but refuse to deliver a 223 byte downlink.

#### Can you do this yourself and submit a Pull Request?
No - I'm not confident I've identified the real location of the bug.

johanstokking, 2021-03-18:
@jamesl-dm thanks for reporting. This seems like a valid issue to me.

rvolosatovs, 2021-03-22:
You are absolutely right, @jamesl-dm, thanks - opened a fix PR at #3966!

closed 2021-03-23 by rvolosatovs