TheThingsNetwork/lorawan-stack #3940: Confirmed uplinks aren't acknowledged while the downlink queue is blocked

jamesl-dm, 2021-03-17:
#### Summary
When there is a large downlink in the downlink queue, which cannot be sent at the current data rate, it stays in the queue until the end node increases data rate, and the downlink becomes deliverable. During this time:

- MAC commands addressed to the end node, like LinkADRReq's to increase its data rate, work as expected
- However, if the device sends a confirmed uplink, the network will NOT send an acknowledgement, which seems wrong

I suspect that a downlink is being prepared with the acknowledgement bit set, and the queued downlink payload. And then that downlink is being deferred because it is too large. The expected behaviour is to send an empty downlink with the ACK bit set instead.

#### Steps to Reproduce
1. Disable ADR on an end node, and fix it to a high spreading factor
2. Queue a large downlink, ie. 100 bytes, that cannot be downlinked at the selected spreading factor
3. Then send a confirmed uplink from the end node

#### What do you see now?
The confirmed uplink is ignored.

#### What do you want to see instead?
An empty downlink with the ACK bit set.

#### Environment
I've tested in AU915.

#### How do you propose to implement this?
No idea...

#### How do you propose to test this?
By verifying that the ACK is sent, even if the downlink queue is blocked.

#### Can you do this yourself and submit a Pull Request?
No

adriansmares, 2021-09-16:
@johanstokking what should we do in these cases where the frame is too big - drop the downlink or stall the queue ?

johanstokking, 2021-09-23:
> @johanstokking what should we do in these cases where the frame is too big - drop the downlink or stall the queue ? (we need to send the ACK always, but just as a general question)

Then fail that downlink message.

closed 2022-08-18 by adriansmares