TheThingsNetwork/lorawan-stack #3944: Timestamp in linux format...

trlafleur, 2021-03-18:
#### Summary

Rx timestamps appear to come for the gateway, some gateway appears to use a non standard format
Would be nice to have a timestamp from the Application Server or other component that has a stable time format.

If you see below, time appears to be always correct, but the timestamp format from the gateway concentrator is NOT reliable...
~~~
 "time": "2020-02-12T15:15:45.787Z",    // ISO 8601 UTC timestamp at which the uplink has been received by the gateway
  "timestamp": 2463457000,               // Timestamp of the gateway concentrator when the message has been received
...
~~~

#### Why do we need this?

...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2021-03-18:
@trlafleur the timestamp there is in fact the concentrator timestamp; the time in microseconds since the gateway concentrator started. It's a 32-bit unsigned int and so it rolls over roughly every 71 minutes. It's not really meaningful on the application layer.

`time` is only present if the gateway has an absolute time, i.e. if there's GPS or NTP.

What's always there and what's pretty reliable is `received_at`.

closed 2021-03-18 by johanstokking

trlafleur, 2021-03-20:
received_at is a Ascii Text string, this is very diffacult to import into
Grafana/inFlux..
would be nice to see this as *numerical value* in linux or other format....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

johanstokking, 2021-03-22:
I understand, but it's not just an ASCII text string, it's strictly following [RFC 3339](https://tools.ietf.org/html/rfc3339) which is the de facto standard for timestamps on the internet. Influx supports RFC 3339 natively; use precision `rfc3339`.