TheThingsNetwork/lorawan-stack #3945: Remove titles for checkbox inputs

kschiffer, 2021-03-18:
#### Summary
We moved to a vertical form layout via #3529. With this, it's no longer necessary to use titles for checkboxes but to only use the label.
#### Why do we need this?
This will reduce the cognitive load of forms and also shrink them in size.

#### What is already there? What do you see now?
Checkboxes like these:
[image]

#### What is missing? What do you want to see?
Checkboxes like these:
[image]

#### How do you propose to implement this?
Trivial. Mostly this comes down to replacing the `title` prop with the `label` prop. However the wording needs to be double checked. It needs to be in imperative tense and describe what the result of checking the checkbox is. To help with this, take the following sentence and complete it with the checkbox label:
`When this option is enabled, we will …`
e.g. `When this option is enabled, we will -> Enforce duty cycles`

#### How do you propose to test this?
Not necessary, this is not a functional change. We'll likely have to update some of the visual e2e tests with new labels.

#### Can you do this yourself and submit a Pull Request?
@asmulko please take the lead.

kschiffer, 2021-03-18:
While at it, please also check the descriptions of checkbox inputs. They are sometimes misleading, e.g. 

[image]

This could be missunderstood in a way that this is already activated. This should rather be `Controls whether the status of this gateway may be publicly displayed`. I think there are more of such issues throughout our forms.

kschiffer, 2021-07-21:
Note that wrt to tooltips, this also means that since tooltips are usually displayed next to the field title, that we need to add a way for tooltips to be displayed next to the checkbox label instead.

NicolasMrad, 2021-09-16:
this has to be done individually while updating the forms. it cannot be generalised that easy, as mentioned by @kschiffer

closed 2021-09-16 by NicolasMrad