TheThingsNetwork/lorawan-stack #3947: New gateway registration flow (Console)

kschiffer, 2021-03-18:
#### Summary
Congruently to #3522, we also want to revisit the gateway registration flow, which is currently too verbose and has many options that are usually not relevant during registration. Additionally, we can use templates of the device repository once we add support for gateways.

#### Why do we need this?
Gateway onboarding is currently not smooth.

#### What is already there? What do you see now?
Gateway registration using the gateway registration form.

#### What is missing? What do you want to see?
- Many UX improvements
  - Remove irrelevant fields
  - Revisit use flow 
  - #3823 
- Onboarding assistance from the device repository

#### How do you propose to implement this?
This issue is currently still a bit of a stub. I will follow up timely with wireframes to make this issue actionable. Implementation will likely happen in two stages:
1. Clean up of the current onboarding without device repository
2. Addition of onboarding via device repository to provide two onboarding modes, congruently with end device onboarding

#### How do you propose to test this?
While doing so, we should also extend the gateway end-to-end tests accordingly.

#### Can you do this yourself and submit a Pull Request?
I'll follow up with the wireframes and then we can decide who picks up the implementation.

johanstokking, 2021-07-13:
> * Clean up of the current onboarding without device repository

Let's start with this indeed.

Do you need a list of fields that are advanced, with their default values? That would eliminate already some fields that are always shown now.

johanstokking, 2021-07-13:
Other things in scope:

- Ability to set antenna placement (indoor or outdoor) (refs https://github.com/TheThingsNetwork/lorawan-stack/issues/4324)
- Filter frequency plans (refs https://github.com/TheThingsNetwork/lorawan-stack/issues/3823, probably needs gateways in device repository)

kschiffer, 2022-05-18:
I have finalized the new UX for gateway registration:
[220517_integrated_gateway_onboarding_01.pdf](https://github.com/TheThingsNetwork/lorawan-stack/files/8715860/220517_integrated_gateway_onboarding_01.pdf)

This also includes planning for a new frequency plan selection experience that will show recommended plans first:
https://github.com/TheThingsNetwork/lorawan-frequency-plans/issues/51

We can now already coordinate implementation of the general restructure of the gateway onboarding.

johanstokking, 2022-05-20:
Should we triage this then? @NicolasMrad

kschiffer, 2022-06-01:
@mjamescompton

Here's the scaffold I would suggest for this:

[image]

Let me know if you have any questions. You can also refer to [my comment on end device onboarding](https://github.com/TheThingsNetwork/lorawan-stack/issues/4847#issuecomment-1143399424). Many notes are relevant for this as well.

johanstokking, 2022-06-04:
References also https://github.com/TheThingsNetwork/lorawan-devices/issues/124.

If there's anything needed that is generic for a particular gateway and model, we might want to load that from the device repository too.

kschiffer, 2022-08-23:
The new gateway onboarding foundation (as laid out here: https://github.com/TheThingsNetwork/lorawan-stack/issues/3947#issuecomment-1143492157) was just merged and is now available in OS and Enterprise.

I'm leaving this issue open since there are still some missing bits:

- https://github.com/TheThingsNetwork/lorawan-stack/issues/3823
- Extending the device repository to include gateways, and making that available through the Console https://github.com/TheThingsNetwork/lorawan-devices/issues/124

@mjamescompton is currently working on updating the documentation accordingly.

KrishnaIyer, 2023-07-04:
The two remaining items here depend on https://github.com/TheThingsIndustries/lorawan-stack/issues/2018. I'm removing the milestone on this until we resolve the blocker.

kschiffer, 2023-07-04:
I would even suggest to close this and track the rest of the progress in #2018 and a new issue once this is resolved.

closed 2023-07-04 by KrishnaIyer