TheThingsNetwork/lorawan-stack #3963: Data Storage integration setup fails in 3.11.3

HeadBoffin, 2021-03-20:
#### Summary
When installing the community edition of the stack, the Data Storage setup fails 

#### Steps to Reproduce
Setup configuration as per docs

Run `docker-compose run --rm stack storage-db init`

Observe:
```
unknown command "storage-db" for "ttn-lw-stack"
ERROR: 255
```

#### What do you want to see instead?
Success

benolayinka, 2021-03-22:
Have another report of this as well @neoaggelos

neoaggelos, 2021-03-22:
The implementation of the Storage Integration is not open-source. As per the [documentation](https://www.thethingsindustries.com/docs/integrations/storage/), the Storage Integration is (presently) not available for self-hosted The Things Stack deployments.

Note that the storage integration is also available in [The Things Network V3](https://eu1.cloud.thethings.network/)

nejraselimovic, 2021-03-22:
I'm closing this as it is apparently working as intended.

closed 2021-03-22 by nejraselimovic

HeadBoffin, 2021-03-25:
> I'm closing this as it is apparently working as intended.

A new world of software development where an error in both documentation AND at the command line is "working as intended".

If it's not available for open-source, perhaps you'd like to

- Change the documentation
- Provide an alternative error message - like 'enterprise license key required'

KrishnaIyer, 2021-03-25:
https://www.thethingsindustries.com/docs/getting-started/installation/running-the-stack/#initialization
```
For the Storage Integration available in The Things Stack Enterprise, the database of the Application Server needs to be initialized as well:

$ docker-compose run --rm stack storage-db init
```

The documentation clearly states that this feature is `available in The Things Stack Enterprise`. What's the issue?

nejraselimovic, 2021-03-25:
> > I'm closing this as it is apparently working as intended.
> 
> A new world of software development where an error in both documentation AND at the command line is "working as intended".
> 
> If it's not available for open-source, perhaps you'd like to
> 
>     * Change the documentation
>       and
> 
>     * Provide an alternative error message - like 'enterprise license key required'

It's not supposed to work for the Open Source and it's documented [here](https://www.thethingsindustries.com/docs/integrations/storage/) as well, so it's not the error in the documentation, neither in the stack itself. 

We can improve the error message indeed, I will create a new issue for that. Thank you for your contribution.

HeadBoffin, 2021-03-25:
> The documentation clearly states that this feature is `available in The Things Stack Enterprise`. What's the issue?

The issue is that I'm reading the documentation with fresh eyes, figuring out the implied bits and trying to make sense of it all. Who actually knows what "The Things Stack Enterprise" means, because you've enabled Data Storage for the TTN v3 servers and in it's current incarnation on v2 it's useful but with a 7 day limit and no meta-data I only prescribe it to clients as a backup, not the main feed. I can't see any mention of how long the data will be held for, so anyone with familiarity with v2 will assume v3 is on 7 days as well.

> It's not supposed to work for the Open Source and it's documented [here]

I've read the page on the link a few times and I can't see where it says "Data Storage is not available in the self-hosted Open Source version" or any reference to any requirements like "requires a purchased Enterprise license".

If it's not explicitly said, we go off past experience and some inference based on what we read. I found this with the Stack install docs, after multiple attempts I got an install down to 10 minutes I was repeating it so often. It's working now and Ben has my notes, but the size of the Red Herring about running it as localhost was so huge that it took some serious hacking to get past the login page when running on an internal network. It was only when I deployed it on a public facing server that the penny dropped that that was what Ben was doing on in his 6 minute install video.

The documentation feels clunky due to the split between the Getting Started which feels like it has some of the info and the Integrations which has more of it and then you have to figure out which bit to cross-reference in the Reference section. Which makes telling forum users where to go to resolve questions that I know are answered in the docs very bitty. I'm building basic examples and sharing them but I find I can end a coding session with about 10 tabs of the documentation open in my browser.