TheThingsNetwork/lorawan-stack #3964: Discover cluster peers with ttn-lw-cli use command

neoaggelos, 2021-03-22:
#### Summary

Update `ttn-lw-cli use <deployment>` command to use the correct server address for all cluster roles (mainly Identity Server). 

#### Why do we need this?

PCN v3 has a single Identity Server (eu1.cloud.thethings.network) and multiple routing clusters (au1, nam1).

#### What is already there? What do you see now?

When we do `ttn-lw-cli use au1.cloud.thethings.network`, the CLI sets `IdentityServerGRPCAddress: au1.cloud.thethings.network`, which is not correct.

#### What is missing? What do you want to see?

Set the correct Identity Server address (also for other components)

 
#### Environment

v3.11.3

#### How do you propose to implement this?

Use [service discovery](https://www.thethingsindustries.com/docs/reference/networking/#service-discovery). If no records are found, fallback to the default host name.

#### How do you propose to test this?

Test locally

#### Can you do this yourself and submit a Pull Request?

Yes

neoaggelos, 2021-03-22:
This requires that SRV records are correctly set for PCN and cloud deployments.

htdvisser, 2021-03-23:
@neoaggelos please assign who you need in this discussion, and what exactly you want to discuss.

closed 2021-04-06 by neoaggelos