TheThingsNetwork/lorawan-stack #3965: ERROR Could not start gRPC server  ---AND--- open /run/secrets/key.pem: permission denied

soportekenai, 2021-03-22:
#### Summary
ERROR Could not start gRPC server 
open /run/secrets/key.pem: permission denied
...

#### Steps to Reproduce

1. I create a VM on Google Cloud (Ubuntu 20.04 LTS)
2. I followed "Installing The Things Stack on Localhost" on Youtube
3. When i run the command docker-compose up, i am getting the following error

the-things-stack_cockroach_1 is up-to-date
the-things-stack_redis_1 is up-to-date
Creating the-things-stack_stack_1 ... done
Attaching to the-things-stack_cockroach_1, the-things-stack_redis_1, the-things-stack_stack_1
cockroach_1  | *
cockroach_1  | * WARNING: ALL SECURITY CONTROLS HAVE BEEN DISABLED!
cockroach_1  | * 
cockroach_1  | * This mode is intended for non-production testing only.
cockroach_1  | * 
cockroach_1  | * In this mode:
cockroach_1  | * - Your cluster is open to any client that can access any of your IP addresses.
cockroach_1  | * - Intruders with access to your machine or network can observe client-server traffic.
cockroach_1  | * - Intruders can log in without password and read or write any data in the cluster.
cockroach_1  | * - Intruders can consume all your server's resources and cause unavailability.
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * INFO: To start a secure server without mandating TLS for clients,
cockroach_1  | * consider --accept-sql-without-tls instead. For other options, see:
cockroach_1  | * 
cockroach_1  | * - https://go.crdb.dev/issue-v/53404/v20.2
cockroach_1  | * - https://www.cockroachlabs.com/docs/v20.2/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "1bc461a070f2" to other nodes, is this routable?
cockroach_1  | * 
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | * 
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * INFO: Replication was disabled for this cluster.
cockroach_1  | * When/if adding nodes in the future, update zone configurations to increase the replication factor.
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2021-03-22 11:11:54.915469294 +0000 UTC (took 5.2s)
cockroach_1  | build:               CCL v20.2.6 @ 2021/03/15 16:04:08 (go1.13.14)
cockroach_1  | webui:               http://1bc461a070f2:26256
cockroach_1  | sql:                 postgresql://root@1bc461a070f2:26257?sslmode=disable
cockroach_1  | RPC client flags:    /cockroach/cockroach <client cmd> --host=1bc461a070f2:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp859079375
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | storage engine:      pebble
cockroach_1  | status:              initialized new cluster
cockroach_1  | clusterID:           09a35609-d628-42c2-b256-91c1f192f89e
cockroach_1  | nodeID:              1
redis_1      | 1:C 22 Mar 2021 11:11:41.782 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 22 Mar 2021 11:11:41.782 # Redis version=6.2.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 22 Mar 2021 11:11:41.782 # Configuration loaded
redis_1      | 1:M 22 Mar 2021 11:11:41.783 * monotonic clock: POSIX clock_gettime
redis_1      | 1:M 22 Mar 2021 11:11:41.786 * Running mode=standalone, port=6379.
redis_1      | 1:M 22 Mar 2021 11:11:41.786 # Server initialized
redis_1      | 1:M 22 Mar 2021 11:11:41.787 * Ready to accept connections
stack_1      |   INFO Setting up core component               
stack_1      |   WARN No cookie hash key configured, generated a random one
stack_1      |   WARN No cookie block key configured, generated a random one
stack_1      |   INFO Setting up Identity Server              
stack_1      |   INFO Setting up Gateway Server               
stack_1      |   INFO Setting up Network Server               
stack_1      |   INFO Setting up Application Server           
stack_1      |   INFO Setting up Join Server                  
stack_1      |   INFO Setting up Console                      
stack_1      |   INFO Setting up Gateway Configuration Server 
stack_1      |   INFO Setting up Device Template Converter    
stack_1      |   INFO Setting up QR Code Generator            
stack_1      |   INFO Setting up Packet Broker Agent          
stack_1      |   INFO Setting up Device Repository            
stack_1      |   INFO Starting...                             
stack_1      |   WARN No cluster key configured, generated a random one key=af0302ea7a2adf0d6
8fc7986350d15dc9cc75fbc30a3ff6d371ac7cfc12900e1 namespace=cluster
stack_1      |   INFO Listening for connections                address=:1884 namespace=grpc p
rotocol=gRPC
stack_1      |  ERROR Could not start gRPC server              error=error:pkg/component:list
ener (could not create `gRPC/tls` listener) error_cause=open /run/secrets/key.pem: permission
 denied protocol=gRPC/tls
stack_1      |   WARN [transport]transport: http2Server.HandleStreams failed to read frame: i
o: read/write on closed pipe namespace=grpc
stack_1      | error:pkg/component:listener (could not create `gRPC/tls` listener)
stack_1      |     protocol=gRPC/tls
stack_1      | --- open /run/secrets/key.pem: permission denied

[image]

[image]

[image]

#### Environment
VM Google Cloud OS UBUNTU 20.04 LTS

nejraselimovic, 2021-03-22:
@soportekenai can you try with this:

`$ sudo chown 886:886 ./cert.pem ./key.pem`

soportekenai, 2021-03-23:
Hello, I no longer have the problem, but now this happens to me

[image]

soportekenai@instance-2:~$ cd the-things-stack
soportekenai@instance-2:~/the-things-stack$ ls
acme  ca-key.pem  ca.json  cert.csr   cert.pem  docker-compose.yml
blob  ca.csr      ca.pem   cert.json  config    key.pem
soportekenai@instance-2:~/the-things-stack$ sudo chown 886:886 ./cert.pem ./key.pem
soportekenai@instance-2:~/the-things-stack$ docker-compose up
Starting the-things-stack_cockroach_1 ... done
Starting the-things-stack_redis_1     ... done
Starting the-things-stack_stack_1     ... done
Attaching to the-things-stack_redis_1, the-things-stack_cockroach_1, the-things-stack_stack_1
redis_1      | 1:C 23 Mar 2021 08:10:22.502 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 23 Mar 2021 08:10:22.502 # Redis version=6.2.1, bits=64, commit=00000000, 
modified=0, pid=1, just started
redis_1      | 1:C 23 Mar 2021 08:10:22.502 # Configuration loaded
cockroach_1  | *
cockroach_1  | * WARNING: ALL SECURITY CONTROLS HAVE BEEN DISABLED!
cockroach_1  | * 
cockroach_1  | * This mode is intended for non-production testing only.
cockroach_1  | * 
cockroach_1  | * In this mode:
cockroach_1  | * - Your cluster is open to any client that can access any of your IP addresse
s.
cockroach_1  | * - Intruders with access to your machine or network can observe client-server
 traffic.
cockroach_1  | * - Intruders can log in without password and read or write any data in the cl
uster.
cockroach_1  | * - Intruders can consume all your server's resources and cause unavailability
.
cockroach_1  | *
redis_1      | 1:M 23 Mar 2021 08:10:22.503 * monotonic clock: POSIX clock_gettime
redis_1      | 1:M 23 Mar 2021 08:10:22.506 * Running mode=standalone, port=6379.
redis_1      | 1:M 23 Mar 2021 08:10:22.506 # Server initialized
redis_1      | 1:M 23 Mar 2021 08:10:22.510 * DB loaded from append only file: 0.002 seconds
redis_1      | 1:M 23 Mar 2021 08:10:22.510 * Ready to accept connections
cockroach_1  | *
cockroach_1  | * INFO: To start a secure server without mandating TLS for clients,
cockroach_1  | * consider --accept-sql-without-tls instead. For other options, see:
cockroach_1  | * 
cockroach_1  | * - https://go.crdb.dev/issue-v/53404/v20.2
cockroach_1  | * - https://www.cockroachlabs.com/docs/v20.2/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "1bc461a070f2" to other nodes, is this routable?
cockroach_1  | * 
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | * 
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2021-03-23 08:10:23.473071259 +0000 UTC (took 0.4s)
cockroach_1  | build:               CCL v20.2.6 @ 2021/03/15 16:04:08 (go1.13.14)
cockroach_1  | webui:               http://1bc461a070f2:26256
cockroach_1  | sql:                 postgresql://root@1bc461a070f2:26257?sslmode=disable
cockroach_1  | RPC client flags:    /cockroach/cockroach <client cmd> --host=1bc461a070f2:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp083194983
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | storage engine:      pebble
cockroach_1  | status:              restarted pre-existing node
cockroach_1  | clusterID:           09a35609-d628-42c2-b256-91c1f192f89e
cockroach_1  | nodeID:              1
stack_1      |   INFO Setting up core component               
stack_1      |   WARN No cookie hash key configured, generated a random one
stack_1      |   WARN No cookie block key configured, generated a random one
[... 27 lines cut ...]
stack_1      |   INFO Request handled                          duration=500µs http.method=GET http.path=/console/login/ttn-stack http.status=302 namespace=web peer.address=176.85.79.1:33686 peer.real_ip=176.85.79.1 request_id=01F1F25MNEWY77YTE2M2ATN17Y
stack_1      |   INFO Request handled                          duration=2.3ms http.method=GET http.path=/assets/materialicons.13a768c54630377e6c74a51096c9c6ef.woff2 http.status=200 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25MNPRX90FJR92FE0VJNP
stack_1      |   INFO Request handled                          duration=2.1ms http.method=GET http.path=/assets/source-sans-pro-v13-latin_latin-ext-regular.d683bfb394f888c5820ce11fc4770fd8.woff http.status=200 namespace=web peer.address=176.85.79.1:33688 peer.real_ip=176.85.79.1 request_id=01F1F25MP7S1T4JDEAMV1JRENX
stack_1      |   INFO Request handled                          duration=3.8ms http.method=GET http.path=/oauth/authorize http.status=302 namespace=web peer.address=176.85.79.1:33689 peer.real_ip=176.85.79.1 request_id=01F1F25MQJZBPYXA3VF5C0KXPH
stack_1      |   INFO Request handled                          duration=900µs http.method=GET http.path=/oauth/login http.status=200 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25MSNM6P9TMBNV8CBBZAE
[... cut ...]
stack_1      |   INFO Request handled                          duration=32.1ms http.method=GET http.path=/oauth/authorize http.status=302 namespace=web peer.address=176.85.79.1:33688 peer.real_ip=176.85.79.1 request_id=01F1F26JF9DV0GQW57VPTASAYV
stack_1      |   INFO Client error                             duration=3.2ms http.method=GET http.path=/console/oauth/callback http.status=403 namespace=web peer.address=176.85.79.1:33688 peer.real_ip=176.85.79.1 request_id=01F1F26JHRA8E7YNJWYJWK85QW
stack_1      |   INFO Request handled                          duration=0s http.method=GET http.path=/healthz/live http.status=200 namespace=w

nejraselimovic, 2021-03-23:
> Hello, I no longer have the problem, but now this happens to me
> 
> [image]
> 
> soportekenai@instance-2:~$ cd the-things-stack soportekenai@instance-2:~/the-things-stack$ ls
> acme ca-key.pem ca.json cert.csr cert.pem docker-compose.yml
> blob ca.csr ca.pem cert.json config key.pem
> soportekenai@instance-2:~/the-things-stack$ sudo chown 886:886 ./cert.pem ./key.pem soportekenai@instance-2:~/the-things-stack$ docker-compose up
> Starting the-things-stack_cockroach_1 ... done
> Starting the-things-stack_redis_1 ... done
> Starting the-things-stack_stack_1 ... done
> Attaching to the-things-stack_redis_1, the-things-stack_cockroach_1, the-things-stack_stack_1
> redis_1 | 1:C 23 Mar 2021 08:10:22.502 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
> redis_1 | 1:C 23 Mar 2021 08:10:22.502 # Redis version=6.2.1, bits=64, commit=00000000,
> modified=0, pid=1, just started
> redis_1 | 1:C 23 Mar 2021 08:10:22.502 # Configuration loaded
> cockroach_1 | *
> cockroach_1 | * WARNING: ALL SECURITY CONTROLS HAVE BEEN DISABLED!
> cockroach_1 | *
> cockroach_1 | * This mode is intended for non-production testing only.
> cockroach_1 | *
> cockroach_1 | * In this mode:
> cockroach_1 | * - Your cluster is open to any client that can access any of your IP addresse
> s.
> cockroach_1 | * - Intruders with access to your machine or network can observe client-server
> traffic.
> cockroach_1 | * - Intruders can log in without password and read or write any data in the cl
> uster.
> cockroach_1 | * - Intruders can consume all your server's resources and cause unavailability
> .
> cockroach_1 | *
> redis_1 | 1:M 23 Mar 2021 08:10:22.503 * monotonic clock: POSIX clock_gettime
> redis_1 | 1:M 23 Mar 2021 08:10:22.506 * Running mode=standalone, port=6379.
> redis_1 | 1:M 23 Mar 2021 08:10:22.506 # Server initialized
> redis_1 | 1:M 23 Mar 2021 08:10:22.510 * DB loaded from append only file: 0.002 seconds
> redis_1 | 1:M 23 Mar 2021 08:10:22.510 * Ready to accept connections
> cockroach_1 | *
> cockroach_1 | * INFO: To start a secure server without mandating TLS for clients,
> cockroach_1 | * consider --accept-sql-without-tls instead. For other options, see:
> cockroach_1 | *
> cockroach_1 | * - https://go.crdb.dev/issue-v/53404/v20.2
> cockroach_1 | * - https://www.cockroachlabs.com/docs/v20.2/secure-a-cluster.html
> cockroach_1 | *
> cockroach_1 | *
> cockroach_1 | * WARNING: neither --listen-addr nor --advertise-addr was specified.
> cockroach_1 | * The server will advertise "1bc461a070f2" to other nodes, is this routable?
> cockroach_1 | *
> cockroach_1 | * Consider using:
> cockroach_1 | * - for local-only servers: --listen-addr=localhost
> cockroach_1 | * - for multi-node clusters: --advertise-addr=<host/IP addr>
> cockroach_1 | *
> cockroach_1 | *
> cockroach_1 | CockroachDB node starting at 2021-03-23 08:10:23.473071259 +0000 UTC (took 0.4s)
> cockroach_1 | build: CCL v20.2.6 @ 2021/03/15 16:04:08 (go1.13.14)
> cockroach_1 | webui: http://1bc461a070f2:26256
> cockroach_1 | sql: postgresql://root@1bc461a070f2:26257?sslmode=disable
> cockroach_1 | RPC client flags: /cockroach/cockroach --host=1bc461a070f2:26257 --insecure
> cockroach_1 | logs: /cockroach/cockroach-data/logs
> cockroach_1 | temp dir: /cockroach/cockroach-data/cockroach-temp083194983
> cockroach_1 | external I/O path: /cockroach/cockroach-data/extern
> cockroach_1 | store[0]: path=/cockroach/cockroach-data
> cockroach_1 | storage engine: pebble
> cockroach_1 | status: restarted pre-existing node
> cockroach_1 | clusterID: 09a35609-d628-42c2-b256-91c1f192f89e
> cockroach_1 | nodeID: 1
> stack_1 | INFO Setting up core component
> stack_1 | WARN No cookie hash key configured, generated a random one
> stack_1 | WARN No cookie block key configured, generated a random one
> stack_1 | INFO Setting up Identity Server
> stack_1 | INFO Setting up Gateway Server
[... 29 lines cut ...]
> stack_1 | INFO Request handled duration=900µs http.method=GET http.path=/oauth/login http.status=200 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25MSNM6P9TMBNV8CBBZAE
> stack_1 | INFO Request handled duration=31.4ms http.method=GET http.path=/assets/account.c4add40ee48703b9c8a5.css http.status=200 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25MX2054X55D0BWYH69H1
> stack_1 | INFO Request handled duration=190.7ms http.method=GET http.path=/assets/account.1138afa41a3aadcd4de8.js http.status=200 namespace=web peer.address=176.85.79.1:33688 peer.real_ip=176.85.79.1 request_id=01F1F25MX65ASC4DZGD7W7DQMV
> stack_1 | INFO Client error duration=500µs http.method=GET http.path=/oauth/api/me http.status=401 namespace=web peer.address=176.85.79.1:33688 peer.real_ip=176.85.79.1 request_id=01F1F25NMFCH8J3R2H3ZD2GGXE
> stack_1 | INFO Request handled duration=200µs http.method=GET http.path=/assets/source-sans-pro-v13-latin_latin-ext-regular.80c998aa03640281d556814a14e1d9c8.woff2 http.status=206 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25NMH4BRWHH0VWGMEGRX8
> stack_1 | INFO Client error duration=100µs http.method=GET http.path=/assets/oauth-favicon.svg http.status=404 namespace=web peer.address=176.85.79.1:33687 peer.real_ip=176.85.79.1 request_id=01F1F25NNRXYP80VQ9H8D78DRZ
[... cut ...]
> stack_1 | INFO Request handled duration=0s http.method=GET http.path=/healthz/live http.status=200 namespace=w

Good, I'm closing this issue then. 

Try following [these steps](https://github.com/TheThingsNetwork/lorawan-stack/issues/3273#issuecomment-704266828), as it is already known as a working setup for localhost. Also (if you haven't already) check out the [Running The Things Stack as localhost](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/) section in our documentation.

closed 2021-03-23 by nejraselimovic

soportekenai, 2021-03-23:
Is there some tutorial for the installation in virtual machine of google?

soportekenai, 2021-03-23:
Now the following message appears: 
Forbidden
Client is not authorized to request a token using this method

nejraselimovic, 2021-03-23:
> Is there some tutorial for the installation in virtual machine of google?

Not that I know of. 

> Now the following message appears:
> Forbidden
> Client is not authorized to request a token using this method
 
Please file a separate issue for this error.