TheThingsNetwork/lorawan-stack #3970: Option to configure the cipher suites

i3mSuperMan, 2021-03-23:
#### Summary
An option to configure the cipher suites to be allowed

#### Why do we need this?
To allow only certain cipher suites for improved security

#### What is already there? What do you see now?
- The default list of cipher suites specified in Go runtime is used

#### What is missing? What do you want to see?
- An option to configure certain cipher suites from the default list

#### Environment
TTS v3.11.3

#### How do you propose to implement this?
- Add `tls.cipher-suites` option 

#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No

johanstokking, 2021-10-12:
From the Go 1.17 release notes:

> Cipher suite ordering is now handled entirely by the crypto/tls package. Currently, cipher suites are sorted based on their security, performance, and hardware support taking into account both the local and peer's hardware. The order of the Config.CipherSuites field is now ignored, as well as the Config.PreferServerCipherSuites field. Note that Config.CipherSuites still allows applications to choose what TLS 1.0–1.2 cipher suites to enable

So we _can_ configure the TLS cipher suites, even with TLS 1.2. That means that we can actually implement this issue.

The customer's problem is an internal security scanning tool that figured out that The Things Stack allows a certain cipher suite that this tool deems insecure. Regardless of that this TLS traffic is within their VPC, that the Go implementation of this cipher suite is in fact not really insecure, and that Envoy probably doesn't even use this cipher suite in the first place, the customer is stuck with an internal security audit tool that reports this as an issue and that doesn't look nice and may lead to compliancy issues.

So my proposal is:
1. Support `tls.cipher-suites` as an experimental feature (with the new package?). This is because Go runtime support may be restricted further in the future. Also, with TLS 1.3 this is likely not even applicable anymore
2. Print a log warning when `tls.cipher-suites` is non-empty

htdvisser, 2021-10-13:
As I've commented before in the related issue, I don't think we should do this. I'm afraid that adding a such a config option will just make it easier to misconfigure a deployment, and it's really not our job to provide workarounds for false-positives coming out of some security scanning tool.

But if we really can't say no to this, a `tls.cipher-suites` indeed looks like the best option. There's no point in using the experimental package for this; we can just mark the config option as experimental (or even deprecated).