TheThingsNetwork/lorawan-stack #3974: Login failed An unknown error occurred. Please try again late

soportekenai, 2021-03-24:
#### Summary

Login failed An unknown error occurred. Please try again late

#### Steps to Reproduce

1. I create a VM on Google Cloud (Ubuntu 20.04 LTS)
2. I followed "Installing The Things Stack 
[YML.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/6197578/YML.zip)
on Localhost" on Youtube
3. I opened an issue
https://github.com/TheThingsNetwork/lorawan-stack/issues/3965
4. now I try to access with firefox on windows and:

[image]

[SSH.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6197587/SSH.txt)
[YML.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/6197591/YML.zip)

nejraselimovic, 2021-03-24:
@kschiffer can you help with this? I can later document it in [Troubleshooting](https://www.thethingsindustries.com/docs/getting-started/installation/troubleshooting/).

kschiffer, 2021-03-26:
Hey @soportekenai, please make sure you have setup the client secret of the console correctly.

You have to set the client secret two times:

1.  As stack configuration (which I see you did correctly)
2.  When creating the console client in the database (you can see that being done in the instruction video here: [https://youtu.be/Owm5IUtQTx8?t=265](https://youtu.be/Owm5IUtQTx8?t=265))

Make sure you set up the correct client secret, so that both values you setup match. In the getting started video, `console` is used as a secret)

nejraselimovic, 2021-03-30:
I followed up on this with a documentation issue to add this error to troubleshooting so I'm going to close this issue. 

@soportekenai if by any case your problem persists, please reopen the issue.

closed 2021-03-30 by nejraselimovic