TheThingsNetwork/lorawan-stack #3991: Preset rights addition to rights setting in console

pe1mew, 2021-03-28:
#### Summary
Add profiles for preset rights in console at collaborator rights for individual users or organizations. 

#### Why do we need this?
The request is to simplify using the existing right-structure available in console to make configuring read-only easier. Therefor it is requested to add: 
- Grant read-only rights

#### What is already there? What do you see now?
In Console currently, predefined settings are available for:
- Grant all current and future rights
- Grant individual rights

#### What is missing? What do you want to see?
As simple method to add read-only rights.

#### How do you propose to implement this?
Add "read-only rights" to the preset options.

#### How do you propose to test this?
Verify if no other rights have been assigned.

#### Can you do this yourself and submit a Pull Request?
NO

kschiffer, 2021-03-30:
Thanks for filing this issue @pe1mew !

Indeed, the simplest way to resolve this is to add another radio button:
[image]

This option then represents all rights that are read-only permissions.

kschiffer, 2021-04-20:
Hello @pe1mew,

After looking into this solution, we determined that using an arbitrary read-only collection of rights is not the way to go. This is because they can give a false impression of security. E.g. granting read only rights would also include the right to read device keys, which is quite intrusive and likely not what you have in mind when you choose to grant read-only rights.

As such we need to look a bit deeper into this and likely come up with a way to save a set of rights as a preset which can then be reused. Doing so will require quite a bit of UX planning as well as implementation. As such, and considering that this would not be a critical feature, this is not something we consider to implement in the short term.

I'll update the issue milestone accordingly. Of course until then, we're still looking forward to your input and ideas about this issue.

bafonins, 2021-04-21:
Related issue on api key rights: https://github.com/TheThingsNetwork/lorawan-stack/issues/3153