TheThingsNetwork/lorawan-stack #3994: Allow `Gs.GetGatewayConnectionStats` when `status_public == true`

htdvisser, 2021-03-29:
#### Summary

The Gateway Server should allow any (authenticated) user to call the `Gs.GetGatewayConnectionStats` RPC when the gateway has `status_public` set to `true`.

#### What is already there? What do you see now?

The Gateway Server does a `rights.RequireGateway(ctx, *ids, ttnpb.RIGHT_GATEWAY_STATUS_READ)` check, which calls the `GatewayAccess.ListRights` RPC on the Identity Server (if the caller provides authentication). The Identity Server then checks membership rights, and returns those.

#### What is missing? What do you want to see?

If the caller is successfully authenticated, but doesn't have `RIGHT_GATEWAY_STATUS_READ`, the Identity Server should doe an additional check of the `status_public` field of the gateway, and add that right if `status_public` is `true`.

#### How do you propose to implement this?

This should probably be implemented in the `GatewayRights` method in `pkg/identityserver/rights.go`.

#### How do you propose to test this?

This can be tested with unit tests.

descartes, 2021-03-29:
For clarity, does this mean the user will have to have some form of authentication in place that is relaxed? (Thereby keeping requests to registered users of TTN)

Or will they be able to omit the authentication headers for anonymous queries?

I'm quite happy with the first, that way TTI can track any frequent requesters.

A daily (or even just weekly) dump of the whole lot would enable those who like stats to get info all in one bite and keep the load off the server but I guess that would be a new issue. Let's get the basic debug going first!

htdvisser, 2021-03-29:
The first indeed. Callers would still need to set an `Authorization: Bearer XXX` header with a valid OAuth Access Token or API key.

PR #4227 (2021-06-02): Allow access to gateway status and location when public

#### Summary

This PR makes changes in the rights API of the Identity Server to return the `RIGHT_GATEWAY_STATUS_READ` and `RIGHT_GATEWAY_LOCATION_READ` rights for authenticated users on gateways that have `status_public` and `location_public` respectively.

Closes #3994

#### Changes

- See above
- Additionally: Make rights functions for other entities consistent

#### Testing

- Updated existing unit tests
- 🐒 

##### Regressions

API consumers that relied on "caller has _any_ rights on this gateway" to show/hide functionality may brea [... cut]
closed 2021-06-02 by htdvisser