TheThingsNetwork/lorawan-stack #3996: Basic Station: after reconnecting to TTN the downlinks aren't processed

forGGe, 2021-03-29:
#### Summary

Sometimes, when the GW changes a network interface preference (say, from WiFi to Cellular), the Basic Station client re-connects to TTN, and downlinks no longer received. Uplinks are working fine.

This was observed on the customer site when running a proprietary BSTN client. Running a packet capture alongside shown that, indeed TTN server doesn't send any downlinks.

#### Steps to Reproduce

Since the BSTN client used for tests is proprietary, the steps described here are informal.

0. Build a custom BSTN client for use with a Linux host.
1. Configure Wireshark to intercept the TLS connection. See here: (section Using LD_PRELOAD) https://security.stackexchange.com/questions/80158/extract-pre-master-keys-from-an-openssl-application and here https://majornetwork.net/2020/02/how-to-decrypt-tls-traffic-with-wireshark/
1. Connect the Linux host to both WiFi and USB modem.
2. Turned on WiFi and turned off USB via console.
3. Start BSTN on Linux host and used LD_PRELOAD trick described above. BSTN connected to TTN via WiFi.
4. Turn off WiFi and turn on the USB modem. Downlinks stopped working.
5. After some time, a BSTN failover kicked-in, and it switched from WiFi to USB. Downlinks began working as usual.
6. Turn off the USB modem and turn on WiFi. Downlinks stopped working again.
7.After some time, the BSTN failover kicked-in, and it switched from USB to WiFi. Downlinks still weren't working, and Wireshark confirmed that no packets were sent by the server (see screenshots and dump).

#### What do you see now?

* TTN displayed "Link ADR request enqueued" message in a device console every time uplink received, or downlink was sent. (see screenshot [1])
* Uplinks were recognized fine. Wireshark confirmed that the client (BSTN) sent some data to the server (TTN), but no responses were delivered. See [2].
* After BSTN is restarted, TTN displayed "Lose downlink data queue," and downlinks started working fine. See [3].
* Wireshark also reported that downlinks are delivered after the restart. See  [4].

[1] TTN logs with ADR message:
[image]

[2] Wireshark: no downlinks, only uplinks (encrypted):
[image]

[3] BSTN restart from the TTN perspective:
[image]

[4] Wireshark, data after the manual restart, downlinks are shown:
[image]

To reinforce my investigation, I'm attaching the following files:

* [capture.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/6222229/capture.zip) - wireshark capture of the communication
  "tcp.stream eq 1" filter will show downlinks not appearing, "tcp.stream eq 6" will show downlinks appearing after the restart.

* [premaster_keys_bstn_dnmsg.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6222225/premaster_keys_bstn_dnmsg.txt) - Pre-master keys to use with the Wireshark and decode server messages (asymmetric encryption, unfortunately, prevents me from seeing client messages though).

* [link_adr_ttn.log](https://github.com/TheThingsNetwork/lorawan-stack/files/6222224/link_adr_ttn.log) - "Link ADR request enqueued" message details.

* [bstn_host_no_dnmsg.log](https://github.com/TheThingsNetwork/lorawan-stack/files/6222223/bstn_host_no_dnmsg.log) - Log of the BSTN after the interface change but before the manual restart (downlinks aren't received).

* [bstn_host_dnmsg_after_restart.log](https://github.com/TheThingsNetwork/lorawan-stack/files/6222222/bstn_host_dnmsg_after_restart.log) - Log of the BSTN after the restart (downlinks are working fine)

#### What do you want to see instead?

Downlinks should be working fine after the network change.

#### Environment

The Things Industries v3.11.3, Custom LoRa GW that supports BSTN protocol and two network interfaces.

#### How do you propose to implement this?

Have no idea.

#### How do you propose to test this?

1. Run the basic station client on the GW with 2 network interfaces.
2. Connect the BSTN to TTN, say, using interface 1. 
3. Disable interface 1, enable interface 2.
4. Let BSTN reconnect using the new interface.
5. Repeat a couple of times.
6. Make sure downlinks working fine.

#### Can you do this yourself and submit a Pull Request?

Nope.

johanstokking, 2021-04-06:
Thanks for the extensive report.

Can you provide the gateway event log as well? You can subscribe via `ttn-lw-cli events subscribe --gateway-id <gateway-id>`

@KrishnaIyer what might be happening here is that old connection (with the other backhaul) is closed after the new one's setup. We should have a disconnection mechanism for that, but that may not work cross-GS instance.

KrishnaIyer, 2021-04-06:
Yup that's possible. We do already have a disconnection mechanism where new connections close existing connections but that does not work cross-GS indeed.

johanstokking, 2021-04-06:
We probably need to check the claim atomically when unclaiming.

It's better to leave garbage claims than removing claims of healthy connections.

KrishnaIyer, 2021-04-13:
I tried reproducing this and the only way that the downlinks would not be delivered at all to the new connection is if the original connection is still active. 

> @KrishnaIyer what might be happening here is that old connection (with the other backhaul) is closed after the new one's setup. We should have a disconnection mechanism for that, but that may not work cross-GS instance.

@johanstokking this might be true during the switch interval. Even in the worst case this takes a couple of minutes, only downlinks scheduled during that time will be lost. Eventually, the old connection would be closed and that path would be unclaimed (assuming there are no bugs in the claiming logic).

For all downlinks after this period, the NS will `GetPeer` and will be presented the instance with the new connection (claim).

Note: We could of course add support to keep claims unique across GSes by first unclaiming IDs (of other peer instances) and then Claim, on every new connection, but let's first see if that really is needed.

@forGGe: I'm a bit suspicious about this
```
5.Turn off WiFi and turn on the USB modem. Downlinks stopped working.
6.After some time, a BSTN failover kicked-in, and it switched from WiFi to USB. Downlinks began working as usual.
7.Turn off the USB modem and turn on WiFi. Downlinks stopped working again.
```

> and it switched from WiFi to USB. Downlinks began working as usual.

This is where my suspicion is. The fact that downlinks just start to work when you switch back to WiFi suggests to me that the original WS connection created through the WiFi network interface doesn't get closed when you switch to USB.

So during the time that the USB modem is on, is the WiFi network interface (as seen from your gateway's user space) still active? Do you have some bridge that tunnels traffic between these network interfaces?

johanstokking, 2021-04-13:
> @johanstokking this might be true during the switch interval. Even in the worst case this takes a couple of minutes, only downlinks scheduled during that time will be lost. Eventually, the old [...]

Indeed that is how it is supposed to work. We need to know if that unclaiming happens properly and timely. If not, and the old peer candidate is still out there, it may be used as the peer.

KrishnaIyer, 2021-04-28:
There's no input from @forGGe and since I can't reproduce this with my local setup, I'm closing this. Feel free to reopen if it's reproducible.

closed 2021-04-28 by KrishnaIyer