TheThingsNetwork/lorawan-stack #4009: Device Type or metadata in Payload Formatter API and messages

niclash, 2021-03-31:
#### Summary

...

#### Why do we need this?
In current v3, Payload Formatters of known devices will place a "decoded_payload" into the "uplink_message", but applications that receive this message has to resort to "object pattern matching" to figure out what device type it came from. Such method is somewhat fragile, and it would be much better to get this in the "uplink_message" directly, rather than somewhat harder "lookup from device_id" which I assume is possible.

For unknown device types, the Payload Formatter that one can add in Console can not AFAIK, look up which device or device type that sent the message. This pushes independent payload formatter for each device, which is seemingly wasteful and error-prone.
...

#### What is already there? What do you see now?
In the messages that external applications are subscribed to, there is no information of how the "decoded_payload" was constructed.
In the JavaScript function signature for decodeUplink (et al), there are only the raw bytes, and that makes it nearly impossible to do application level decoding as soon as more than one device type is used, other than putting it into every single device.
...

#### What is missing? What do you want to see?
In messages, I would like to have an additional section "device_info", with "brand", "model", "hw_version" and "fw_version".
```json
{
  "device_info" : {
    "brand" : "dragino",
    "model" : "lht65",
    "hw_version" : "unknown",
    "fw_version" : "1.8"
  }
}
```
...

#### How do you propose to implement this?
I don't (yet) know the internals of the stack and have no opinion.

...

#### How do you propose to test this?
Again, I don't know anything about the stack's internals or how the tests are built up or executed at the moment.
...

#### Can you do this yourself and submit a Pull Request?
Not at the moment.

...

johanstokking, 2021-04-07:
@niclash I understand and this has been raised before. The reason why we don't include this, is to reduce bandwidth and to keep the Identity Server (where this information is stored) out of the hot path (i.e. Application Server would need to get this from IS on each uplink message).

I do understand, however, that from a stateless message process point of view, this is valuable information. Also, AS is now already getting information from IS periodically.

This is a sliding edge though. I can think of:

1. Device brand/model/versions
2. Attributes
3. Last reported location

Since (2) can be arbitrary data that the application layer cannot process in a generic way, and because it can become quite sizable (10 pairs of 36 char key and 200 char value), we decided against this.

The location (3) is already made available as "location changed" events and the application layer is expected to keep track of that.

So I think we can draw a line here and do include a field `EndDeviceVersionIdentifiers`:

https://github.com/TheThingsNetwork/lorawan-stack/blob/3560d869e1f8fa86386320ef15e7529d702d3d32/api/end_device.proto#L135-L142

AS would need to get this from IS. I believe it may already have the information now for getting the payload decoders from the device repository.

niclash, 2021-04-07:
"So I think we can draw a line here and do include a field EndDeviceVersionIdentifiers" --> I think that would be awesome.