TheThingsNetwork/lorawan-stack #4012: Go get fails

aureliar8, 2021-03-31:
#### Summary
I would like to be able to import the package `go.thethings.network/lorawan-stack/v3/pkg/ttnpb` in my go program. 
To do so, I need to be able to `go get`it, but it fails. 

#### Steps to Reproduce
```sh
$ mkdir test && cd test
$ go mod init main 
$ go get -v go.thethings.network/lorawan-stack/v3/pkg/ttnpb
```
This issue can also be triggered by running `go get -v go.thethings.network/lorawan-stack/v3/`
#### What do you see now?
`go get` takes a few minutes then: 
```
get "gopkg.in/DATA-DOG/go-sqlmock.v1": found meta tag vcs.metaImport{Prefix:"gopkg.in/DATA-DOG/go-sqlmock.v1", VCS:"git", RepoRoot:"https://gopkg.in/DATA-DOG/go-sqlmock.v1"} at //gopkg.in/DATA-DOG/go-sqlmock.v1?go-get=1
get "go.thethings.network/lorawan-stack/v3": found meta tag vcs.metaImport{Prefix:"go.thethings.network/lorawan-stack", VCS:"git", RepoRoot:"https://github.com/TheThingsNetwork/lorawan-stack"} at //go.thethings.network/lorawan-stack/v3?go-get=1
get "go.thethings.network/lorawan-stack/v3": verifying non-authoritative meta tag
go get: go.thethings.network/lorawan-stack/v3@none updating to
	go.thethings.network/lorawan-stack/v3@v3.11.3 requires
	gopkg.in/DATA-DOG/go-sqlmock.v1@v1.0.0-00010101000000-000000000000: invalid version: unknown revision 000000000000 
```
#### What do you want to see instead?
Nothing, it should succeed

#### Environment
MacOs: go1.16
Manjaro Linux: go1.14

#### How do you propose to implement this?
In the main `go.mod`there is a line `gopkg.in/DATA-DOG/go-sqlmock.v1 v1.0.0-00010101000000-000000000000 // indirect`. I guess this should be edited ? 

#### How do you propose to test this?
Ensure that `go get -v go.thethings.network/lorawan-stack/v3/pkg/ttnpb` succeeds

#### Can you do this yourself and submit a Pull Request?
Don't think so

nejraselimovic, 2021-04-01:
@rvolosatovs any way you can help with this?

neoaggelos, 2021-04-02:
Hi @aureliar8

You will need to add the following modules in your `go.mod` file:

```
replace gopkg.in/DATA-DOG/go-sqlmock.v1 => gopkg.in/DATA-DOG/go-sqlmock.v1 v1.3.0

replace gocloud.dev => gocloud.dev v0.19.0

replace github.com/grpc-ecosystem/grpc-gateway => github.com/TheThingsIndustries/grpc-gateway v1.15.2-gogo
```

See also https://github.com/TheThingsIndustries/lorawan-stack-examples/blob/go-examples/go/go.mod

aureliar8, 2021-04-02:
Thanks ! Everything works now

closed 2021-04-02 by aureliar8