TheThingsNetwork/lorawan-stack #4014: [Console] Add option for uint32_t format 0x00000000 for Device address field

lnlp, 2021-04-02:
#### Summary
In the Console for a device, for the 'Device address' field add the uint32_t format 0x000000 as option.

In the Console for a device the 'Device address' field can be copied.
In C/C++ code the device address is usually defined as uint32_t type which requires the format 0x000000.
But this format is currently not available as one of the options.

#### Why do we need this?
Users coding in C/C++ cannot use a copy/pasted device address value because it is in the wrong format so they have to manually change the the format of the pasted device address value after each copy/paste action.
This is very impractical. Copy/paste should just work without having to make any manual changes after the paste.

#### What is already there? What do you see now?
By default the device address is shown as a sequence of bytes separated by a space:
[image]

It is also possible to display and copy it in C/C++ msb and lsb formats:
[image]

[image]

#### What is missing? What do you want to see?
Add a format option for uint32_t type with a value looking like: 0x00000000

#### Environment
_Not relevant._

#### How do you propose to implement this?
Add the format option as described.

#### How do you propose to test this?
\-

#### Can you do this yourself and submit a Pull Request?
No.

kschiffer, 2021-04-06:
Thanks for filing this issue @lnlp 

I think what we can do here is to allow a third representation for the format toggle. This would mean that the format toggle would cycle through hex, array, and uint32\_t notation. This third option should be disabled by default, so we can overwrite the setting for the device address only.

lnlp, 2021-04-06:
> This third option should be disabled by default, so we can overwrite the setting for the device address only.

It's not clear what you mean. There is nothing that needs to be _disabled_.  Maybe you just mean that the uint32_t type should not be the format that is shown as default?

> This would mean that the format toggle would cycle through hex, array, and uint32_t notation.

Yes. But there is nothing to '_disable_ by default' here.

kschiffer, 2021-04-06:
Sorry, these were more implementation notes for ourselves.

What I mean is that it should not be possible to toggle into the `uint32_t` notation by default, but only for specific fields, such as the device address.

lnlp, 2021-04-06:
> but only for specific fields, such as the device address.

This relates to the device address only because only the device address fits in a scalar C/C++ data type.
The other keys all require use of an array.

PR #4058 (2021-04-21): Add uint32 notation for device address
#### Summary

Closes #4014 

#### Changes

- Added C-Style uint32_t representation for device address field

Before:
No option

After:
[image]

#### Testing

- Manually tested

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: Relevant documentation is added or updated.
- [x] Changelog: Significa [... cut]
closed 2021-04-21 by asmulko