TheThingsNetwork/lorawan-stack #4016: Simplify CLI configuration for TTN v3 and TTSC

neoaggelos, 2021-04-06:
#### Summary

Simplify CLI configuration for The Things Network V3 and The Things Stack Cloud deployments.

#### Why do we need this?

All TTN clusters share the Identity Server from eu1. This breaks the `ttn-lw-cli use nam1.cloud.thethings.network` command, because it sets the Identity Server address to `nam1.cloud.thethings.network`, which is not correct.

#### What is already there? What do you see now?

`ttn-lw-cli use [cluster]` command can be used to configure the CLI for use with a single-host deployment at `[cluster]`

#### What is missing? What do you want to see?

Make it straightforward to use the `ttn-lw-cli use` command with the TTN and TTSC deployments. 

#### Environment

3.12

#### How do you propose to implement this?

The TTN and TTSC deployments use well-known host addresses, and we do not expect them to change frequently. Ideally, we could SRV records for this (https://github.com/TheThingsNetwork/lorawan-stack/issues/3964), but we can also start with parsing the specified host in the CLI and:

- If we match a `([a-z0-9]+).cloud.thethings.network`, then set Identity Server and OAuth address to `eu1.cloud.thethings.network`.
- If we match a `([a-z0-9\-_]+).([a-z0-9]+).cloud.thethings.industries`, then set Identity Server and OAuth address to `\1.eu1.cloud.thethings.industries`.

If the region does not match any of the known deployments, we could also print a warning message.

#### How do you propose to test this?

Testing locally should be enough

#### Can you do this yourself and submit a Pull Request?

Yes

neoaggelos, 2021-04-06:
Replaces #3964

neoaggelos, 2021-06-21:
My suggestion for this would be to add two new commands (instead of trying hackish overloads of the existing `use` command):

```bash
# For The Things Stack Community Edition ($cluster.cloud.thethings.network)
$ ttn-lw-cli use-community [eu1 nam1 au1]

# For The Things Stack Cloud ($tenant.$cluster.cloud.thethings.industries)
$ ttn-lw-cli use-cloud --cluster [eu1 nam1 au1] --tenant [tenant-id]
```

We can have helpful warning/error messages if people enter something wrong (e.g. have a warning when an unknown cluster name is given, or a tenant id is invalid).

We can then update the documentation with 3 separate tabs: one for TTSCE (showing `use-community`), one for TTSC (showing `use-cloud`), and one for personal deployments (showing `use`, like now). (cc @benolayinka)

The cluster configuration is static enough that we know the identity server and oauth server addresses. When new clusters are deployed, we can simply extend the list of known cluster names.

 I have done some initial work in https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/cli-config-cloud

htdvisser, 2021-07-02:
This did not make it for the v3.13.3 deadline, so pushing to v3.14.0

PR #4315 (2021-07-21): Make CLI configuration for TTS Cloud and TTS Community Edition easier

#### Summary

Closes #4016
Closes #3591

#### Changes

- When passing a TTS Cloud or TTS Community Edition hostname, configure OAuth and Identity Server address.
- Validate hostname passed to `ttn-lw-cli use` command.

#### Testing

```
$ ttn-lw-cli use nam1.cloud.thethings.network
INFO    Configuring for The Things Stack Community Edition      {"cluster_id": "nam1"}
INFO    Set OAuth Server address        {"address": "https://eu1.cloud.thethings.network/oauth"}
INFO    Set Identity Server gRPC address        {"address": "e [... cut]
closed 2021-07-21 by neoaggelos