TheThingsNetwork/lorawan-stack #4018: Store application uplinks as part of the session

adriansmares, 2021-04-06:
#### Summary

The Application Server should store, in a FIFO fashion, the application data uplinks as part of the device session.

#### Why do we need this?

Certain integrations may want to query the last received data uplinks. Since we would like to avoid enabling the storage integration just for this, we could bring back the `recent_uplinks` field of `ttnpb.EndDevice`:

https://github.com/TheThingsNetwork/lorawan-stack/blob/e37d779528c17aabf33d9149c5603e31e9206217/api/end_device.proto#L534

#### What is already there? What do you see now?

The storage integration keeps these uplinks in long term storage.

#### What is missing? What do you want to see?

The Application Server should store, either as part of the end device, or of the session, the last 16 application data uplinks.

#### Environment

`v3.12`

#### How do you propose to implement this?

Add the API field, save the uplinks as part of `as.handleUplink`, and allow frontends to query the recent uplinks via `io.Server`.

#### How do you propose to test this?

Send some real uplinks from and end device and check that they are stored.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @johanstokking

adriansmares, 2021-04-06:
Blocking https://github.com/TheThingsNetwork/lorawan-stack/issues/4019

htdvisser, 2021-04-14:
Since https://github.com/TheThingsNetwork/lorawan-stack/issues/4019 is now planned for 3.12.2, this one can also be moved

PR #4061 (2021-04-21): Store recent data application uplinks in the Application Server 

#### Summary

Closes #4018

#### Changes

- Break down the `io.Server` interface into multiple sub-interfaces, as it was getting too big.
- Add a data application uplink registry. 

#### Testing

Unit testing.

##### Regressions

This PR adds a new store, for which unit tests are added. The usage of the functionality will be tested better as part of https://github.com/TheThingsNetwork/lorawan-stack/issues/4019.

#### Notes for Reviewers

This implementation diverges from the proposal due to the following reasons:
- We always fet [... cut]
closed 2021-04-21 by adriansmares