TheThingsNetwork/lorawan-stack #4019: Multi frame request support for LoRaCloud GLS

adriansmares, 2021-04-06:
#### Summary

References https://github.com/TheThingsNetwork/lorawan-stack/issues/4018

We should support sending multi frame requests via GLS, using the recently sent uplinks.

#### Why do we need this?

In order to have better location queries, since multi frame should provide more accurate results.

#### What is already there? What do you see now?

The `QUERY_TOARSSI` query type, which does a single frame request using the `TOARSSI` algorithm.

#### What is missing? What do you want to see?

1. A new configuration option that should specify if the query should be single frame or multi frame (not a new query type).
2. A new configuration option for how many of the recent frames we should we use for the multi frame queries (if they are enabled). We should also provide a way to automatically determine this information, based on the first byte of the payload.
3. A new configuration option for how recent (a duration) should the recent uplinks be, when used by the multi frame query.
4. On uplink handling, we should check if we should do a single frame request, as we currently do, or use multiple frames using the new configuration.

#### Environment

`v3.12`

#### How do you propose to implement this?

Add the requested configuration to the package (new fields), and then use them as part of the uplink handling.

#### How do you propose to test this?

Send multiple uplinks and check if the query is done. You can use the simulation functionality.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @johanstokking

adriansmares, 2021-04-06:
Blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/4018

htdvisser, 2021-04-14:
Since @adriansmares is off today, and it isn't in progress yet, this isn't going to make today's deadline, so moved to v3.12.2.

closed 2021-04-28 by adriansmares