TheThingsNetwork/lorawan-stack #4028: RX1_DELAY adjustable in console - preferably as global application setting

pomplesiegel, 2021-04-06:
Hello Everyone,

It would be very helpful to adjust the RX1_DELAY globally per application through the v3 console. I understand that a user could potentially get in trouble this way, but if it were under "advanced settings" one could try it at their own risk. I would certainly appreciate the ability to adjust this. 

Thank you!

nejraselimovic, 2021-04-07:
@pomplesiegel thank you for reporting the issue. 

Setting the desired RX1 delay through the Console will be available for OTAA devices too (for ABP devices it is already available). There is already an issue about it: #2047 - you can see it under **Low** priority -> `desired_rx1_delay`. Adjusting this through the Console will be possible and this issue is in progress, but until it's finished, you can do it via CLI. 

We nurture the practice of keeping the application settings related only to the application layer, while setting the RX1 delay is a MAC layer setting. 

@bafonins can answer if you have more questions about this.

pomplesiegel, 2021-04-07:
OK, thank you very much! Is there a way to globally set the desired_rx1_delay for an entire application through the CLI? This is also what I would believe would be helpful in the console version if possible.

nejraselimovic, 2021-04-07:
> OK, thank you very much! Is there a way to globally set the desired_rx1_delay for an entire application through the CLI? This is also what I would believe would be helpful in the console version if possible.

No, MAC settings cannot be set on the application level at all. The Network Server looks at applications only as a security domain, nothing else.

pomplesiegel, 2021-04-07:
OK, understood. Thank you

closed 2021-04-07 by nejraselimovic