TheThingsNetwork/lorawan-stack #4041: MAC commands can't delete MacState channel

aureliar8, 2021-04-08:
#### Summary
The channels a device uses to communicate  can be modified by the network server using MAC commands (NewChannelReq & NewChannelAns). 
This should work to add new channels but also to remove some 

But it seems that removing a channel isn't implemented 

#### Steps to Reproduce

Add this test case to `pkg/networkserver/mac/new_channel_test.go#TestHandleNewChannelAns` : 

```
{
	Name: "Delete channel",
	Device: &ttnpb.EndDevice{
		MACState: &ttnpb.MACState{
			PendingRequests: []*ttnpb.MACCommand{
				(&ttnpb.MACCommand_NewChannelReq{
					ChannelIndex:     4,
					Frequency:        0,
					MinDataRateIndex: 0,
					MaxDataRateIndex: 0,
				}).MACCommand(),
			},
			CurrentParameters: ttnpb.MACParameters{
				Channels: []*ttnpb.MACParameters_Channel{
					nil,
					nil,
					nil,
					nil,
					{
						DownlinkFrequency: 42,
						UplinkFrequency:   42,
						MinDataRateIndex:  2,
						MaxDataRateIndex:  3,
						EnableUplink:      true,
					},
				},
			},
		},
	},
	Expected: &ttnpb.EndDevice{
		MACState: &ttnpb.MACState{
			CurrentParameters: ttnpb.MACParameters{
				Channels: []*ttnpb.MACParameters_Channel{
					nil,
					nil,
					nil,
					nil,
				},
			},
			PendingRequests: []*ttnpb.MACCommand{},
		},
	},
	Payload: &ttnpb.MACCommand_NewChannelAns{
		FrequencyAck: true,
		DataRateAck:  true,
	},
	Events: events.Builders{
		EvtReceiveNewChannelAccept.With(events.WithData(&ttnpb.MACCommand_NewChannelAns{
			FrequencyAck: true,
			DataRateAck:  true,
		})),
	},
}
```

#### What do you see now?
The test fails, the channel is still here and was not removed 

#### What do you want to see instead?
The test suceeds and the channel is removed 
#### Environment
v3.12

#### How do you propose to implement this?
In `pkg/networkserver/mac/new_channel.go#HandleNewChannelAns`, add a case if the frequency requested was 0 (meaning we wanted to delete the channel), and if  `len(DesiredParameters.Channels) < len(CurrentParameters.Channels)`

#### How do you propose to test this?
The test should pass 
#### Can you do this yourself and submit a Pull Request?
No, I don't know how to solve the following issues:

- HandleNewChannelAns can be called multiple times if there is duplicated uplinks, but we should only remove once 
- How to keep chIndex coherent  with other mac commands (that may be in the same uplink) if we want to delete a channel that isn't the last one

rvolosatovs, 2021-04-10:
Thank you for report and especially the test case!

From 1.0.4 specification:
[image]

Network Server [disables](https://github.com/TheThingsNetwork/lorawan-stack/blob/3f55accd8211ab7437d4317813882246dd9d6249/pkg/networkserver/mac/new_channel.go#L183) uplink on the channel, for which a frequency of 0 is sent.

Technically, the channel is still usable for downlink in class B and C, since in class A there is no band, for which `NewChannelReq` is supported with non-symmetric channels if I remember correctly. This is *probably* a bug.

The way I understand *disable* in this context is that it is not usable for uplink/downlink, but still could be enabled for uplink via e.g. `LinkADRReq`

I'm not sure what's the right way to handle this, could you tell more about your use case for context? I.e. why is it desirable to explicitly *remove* the channel and not just disable traffic?

aureliar8, 2021-04-12:
I stumbled upon this issue when a device had a `MACState.CurrentParameters.Channels` of size greater than `MACState.DesiredParameters.Channels`. 

In this case [DeviceNeedsNewChannelReq](https://github.com/TheThingsNetwork/lorawan-stack/blob/3f55accd8211ab7437d4317813882246dd9d6249/pkg/networkserver/mac/new_channel.go#L72) always return true, which will send MACCommands in the next downlink. But because no answer can change that  `len(MACState.DesiredParameters.Channels) > len(MACState.CurrentParameters.Channels)`, it will not only repeat indefinitely but also prevent sending application downlink (as MACCommands take priority)  

However it is not yet clear why the device reached this state in the first place

rvolosatovs, 2021-05-19:
> I stumbled upon this issue when a device had a `MACState.CurrentParameters.Channels` of size greater than `MACState.DesiredParameters.Channels`.
>
> In this case [...]

Brilliant, thanks!

closed 2022-09-20 by adriansmares