TheThingsNetwork/lorawan-stack #4042: CLI fails to import device JSON exported with ttn-lw-migrate

matthijskooijman, 2021-04-08:
#### Summary
I tried exporting a device using ttn-lw-migrate, which produces a JSON file. However, trying to import it [according to docs](https://www.thethingsnetwork.org/docs/the-things-stack/migrate-to-v3/index.html), I get:

```
$ ttn-lw-cli end-devices create --application-id meet-je-stad < device.json 
error:pkg/ttnpb:parse (could not parse `{"value":6}` into `ADRAckLimitExponent`)
    field=ADRAckLimitExponent
    value={"value":6}
--- strconv.ParseInt: parsing "{\"value\":6}": invalid syntax
```

The same JSON file does import successfully through the web console, so I'm assuming the format it uses is correct, but the cli and web console are inconsistent in the format they accept (this is also why I'm reporting this here, rather than at the lorawan-stack-migrate repo).

#### Steps to Reproduce

1. Run this command, on the attached [device.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6280701/device.txt): `ttn-lw-cli end-devices create --application-id meet-je-stad < device.txt`

Note that the device.txt file was created using:

    ttn-lw-migrate device  --ttnv2.app-id meet-je-stad --ttnv2.app-access-key ttn-account-v2.REDACTED --ttnv2.frequency-plan-id EU_863_870_TTN --ttnv2.with-session=false --source ttnv2 meetstation-2022

I redacted the app key in the file, otherwise it is identical to the one produced using the above command (using ttn-lw-migrate 0.4.0-rc1, but I checked git master and it generates the same format).

#### What do you see now?
The error above.

#### What do you want to see instead?
A successful import of the device.

#### Environment
```
matthijs@grubby:~$ ttn-lw-cli version 
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.12.0
Build date:          2021-04-07T07:51:20Z
Git commit:          70859e462
Go version:          go1.16.3
OS/Arch:             linux/amd64
```
Installed from snap.

#### How do you propose to implement this?
...

#### How do you propose to test this?
The attached JSON file could be added as a testcase.

#### Can you do this yourself and submit a Pull Request?
Probably not, no time and not familiar with the codebase.

neoaggelos, 2021-04-09:
Hi @matthijskooijman, thanks for reporting the issue.

The Things Stack 3.12 introduced breaking API changes in the way MAC settings fields are handled, which is why you are seeing the issue.

- There is a PR to update `ttn-lw-migrate` for the new API (see https://github.com/TheThingsNetwork/lorawan-stack-migrate/pull/26), and 0.5 will be released afterwards to include this change.
- In the meantime, you can still import the devices by editing the JSON file by hand and replacing `{"value":6}` with `6`. Similarly for the rest of the fields, replace all `{"value": <VALUE>}` instances with simply `<VALUE>` instead.

neoaggelos, 2021-04-26:
This is now fixed in [ttn-lw-migrate 0.5.0](https://github.com/TheThingsNetwork/lorawan-stack-migrate/releases/tag/v0.5.0)

closed 2021-04-26 by neoaggelos