TheThingsNetwork/lorawan-stack #4051: LoRaWAN version and regional parameters cleared on console

jpmeijers, 2021-04-12:
#### Summary

I registered a device, and I set the LoRaWan version, as well as the regional parameters. Going into the device settings, these fields are not set. I tried setting it again and saving, but navigating away and back into the settings, these values are again cleared.

[image]

#### Steps to Reproduce

1. Register a device and set the lorawan version and regional params version
2. Go into the device network layer settings
3. See that the first two fields are cleared

#### What do you see now?

Fields not set

#### What do you want to see instead?

Fields should be set

#### Environment

TTN v3.12.0

#### How do you propose to implement this?

...

#### How do you propose to test this?

Same as repoduction steps

#### Can you do this yourself and submit a Pull Request?

No

i3mSuperMan, 2021-04-13:
@kschiffer,

Facing the same issue. However, able to fetch the device details via CLI. The issue doesn’t seem to affect uplink/downlink functionality. It could be mostly a Console related issue. But, would like others to confirm on the same.

kschiffer, 2021-04-13:
This is likely fixed by #4026, but need @bafonins to confirm.

bafonins, 2021-04-15:
This will be fixed in the next release, see https://github.com/TheThingsNetwork/lorawan-stack/pull/4026 for more details

closed 2021-04-15 by bafonins