TheThingsNetwork/lorawan-stack #4056: Make gateway & application names searchable

laurensslats, 2021-04-15:
#### Summary
As of now, you can only search on GatewayID or ApplicationID, not on the Name of Gateways of Applications.

#### Why do we need this?
Easy to find gateways or applications

#### What is already there? What do you see now?
Search for GatewayID or Application ID

#### What is missing? What do you want to see?
Search for Gateway name or Application name

#### Environment
V3.12

#### How do you propose to implement this?
Extend searchable fields

kschiffer, 2021-10-08:
Yes, so we currently cannot search by ID **and** name, so the only way forward is implementing an advanced search feature that allows searching by different properties. This can be tracked here: #1825

pgalic96, 2021-11-09:
Closed because separate issues are merged together #1825

closed 2021-11-09 by pgalic96