TheThingsNetwork/lorawan-stack #4063: Centralized API to get the total number of devices in a tenant.

i3mSuperMan, 2021-04-16:
#### Summary
It would be very useful to have a centralized API that returns the total number of devices registered in a tenant.
Refs: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/398

#### Why do we need this?
Users can use this API to monitor the current number of registered devices from time to time.

#### What is already there? What do you see now?
Admin users in a tenant can list down all the applications in a tenant and then walk through each application to count the devices in it.
Refs: https://www.thethingsindustries.com/docs/reference/cli/applications/#command-ttn-lw-cli-applications-search
https://www.thethingsindustries.com/docs/reference/api/application/#the-entityregistrysearch-service

#### What is missing? What do you want to see?
A centralized API that fetches the count of end-devices in a tenant

#### Environment
TTS v3.12.0

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?
No, @adriansmares

nejraselimovic, 2021-04-16:
@adriansmares please reassign if needed.

adriansmares, 2021-04-21:
To give more context on this issue:

It would be useful as a network administrator to be able to obtain an overview of the number of registered end devices in each application. The API proposed here should be available to network administrators and return a mapping between application IDs and the number of registered end devices of each application. 

We can then display this in table form as part of the Console, but that is out of the scope for this issue.

closed 2025-04-30 by KrishnaIyer