TheThingsNetwork/lorawan-stack #4066: LNS Auth field in the Gateway Settings page displays garbage value

KrishnaIyer, 2021-04-16:
#### Summary

LNS Auth field in the Gateway Settings page displays garbage value

#### Steps to Reproduce

1. Update the LBS LNS Key or clear it.
2. Press Save changes

[image]

#### What do you see now?

Some garbage value is printed.

#### What do you want to see instead?

Print the LNS Key value (if set) or be empty.

#### Environment

```
tts v3.12.0
```

#### How do you propose to implement this?

Should be something in the type conversion. This was actually tested in the original PR; https://github.com/TheThingsNetwork/lorawan-stack/pull/4011

But there might have been a regression introduced.

#### How do you propose to test this?

Run the stack and test this.

#### Can you do this yourself and submit a Pull Request?

Yes

KrishnaIyer, 2021-05-20:
Ok so I tried to reproduce this but I can't do this locally but I see it in deployed clusters.

Also the issue is only on the `Save Changes` .

I guess this has something to do with `dev` v/s `prod` environments? @kschiffer any ideas?

bafonins, 2021-06-03:
This does not seem to be Console related:

```
const value = "NNSXS.POHZLANBQCECZHMFLIXOGRBLXYZ5EOTWU3ZDW3I.5RATR3TV7QJGCLMFROODIPQD4WTCPKOVSNVGSN4BNHFYW3G5A3OA"
const b64Value = btoa(value) // "Tk5TWFMuUE9IWkxBTkJRQ0VDWkhNRkxJWE9HUkJMWFlaNUVPVFdVM1pEVzNJLjVSQVRSM1RWN1FKR0NMTUZST09ESVBRRDRXVENQS09WU05WR1NONEJOSEZZVzNHNUEzT0E="

// request:
curl --location --request PUT 'http://localhost:8080/api/v3/gateways/<gtw-id>' \
--header 'Authorization: Bearer ...' \
--header 'Content-Type: application/json' \
--data-raw '{
  "gateway": {
    "lbs_lns_secret": {
      "value": "Tk5TWFMuUE9IWkxBTkJRQ0VDWkhNRkxJWE9HUkJMWFlaNUVPVFdVM1pEVzNJLjVSQVRSM1RWN1FKR0NMTUZST09ESVBRRDRXVENQS09WU05WR1NONEJOSEZZVzNHNUEzT0E="
    }
  },
  "field_mask": {
    "paths": [
      "lbs_lns_secret",
      "lbs_lns_secret.value"
    ]
  }
}'

// response:
{
    "ids": {
        "gateway_id": "test-gateway1",
        "eui": "1237871238912378"
    },
    "created_at": "2021-06-03T11:21:46.797Z",
    "updated_at": "2021-06-03T11:40:21.565Z",
    "version_ids": {},
    "lbs_lns_secret": {
        "value": "Tk5TWFMuUE9IWkxBTkJRQ0VDWkhNRkxJWE9HUkJMWFlaNUVPVFdVM1pEVzNJLjVSQVRSM1RWN1FKR0NMTUZST09ESVBRRDRXVENQS09WU05WR1NONEJOSEZZVzNHNUEzT0E="
    }
}

const returned_secret = atob("Tk5TWFMuUE9IWkxBTkJRQ0VDWkhNRkxJWE9HUkJMWFlaNUVPVFdVM1pEVzNJLjVSQVRSM1RWN1FKR0NMTUZST09ESVBRRDRXVENQS09WU05WR1NONEJOSEZZVzNHNUEzT0E=") // "NNSXS.POHZLANBQCECZHMFLIXOGRBLXYZ5EOTWU3ZDW3I.5RATR3TV7QJGCLMFROODIPQD4WTCPKOVSNVGSN4BNHFYW3G5A3OA"
value === returned_secret // true

// now on staging1:

// request:
curl --location --request PUT 'https://tti.staging1.cloud.thethings.industries/api/v3/gateways/<gtw-id>' \
--header 'Authorization: Bearer ...' \
--header 'Content-Type: application/json' \
--data-raw '{
  "gateway": {
    "lbs_lns_secret": {
      "value": "Tk5TWFMuUE9IWkxBTkJRQ0VDWkhNRkxJWE9HUkJMWFlaNUVPVFdVM1pEVzNJLjVSQVRSM1RWN1FKR0NMTUZST09ESVBRRDRXVENQS09WU05WR1NONEJOSEZZVzNHNUEzT0E="
    }
  },
  "field_mask": {
    "paths": [
      "lbs_lns_secret",
      "lbs_lns_secret.value"
    ]
  }
}'

// response:
{
    "ids": {
        "gateway_id": "bafonins-test-gateway",
        "eui": "1237897123891273"
    },
    "created_at": "2021-05-25T12:48:19.239Z",
    "updated_at": "2021-06-03T11:48:19.460Z",
    "version_ids": {},
    "lbs_lns_secret": {
        "key_id": "is/gateway-secrets-encryption-key",
        "value": "Npp5i8+WTNaiWJXx4xs5mkoX9HFQJ2QQ7mhikGzaZNEKAxjcRzEz0KOMrMi0CALSKCCdQ0pmGiNBcMymoqMLjqIKYxRAb035JpDF29M/vRfgpYQ8wbXwVC1kSgLB5u+1T4Hs5CKD/PVWBJl+/sXRpxXS1zijllZWbDipcMBM"
    }
}

const returned_secret2 = atob("Npp5i8+WTNaiWJXx4xs5mkoX9HFQJ2QQ7mhikGzaZNEKAxjcRzEz0KOMrMi0CALSKCCdQ0pmGiNBcMymoqMLjqIKYxRAb035JpDF29M/vRfgpYQ8wbXwVC1kSgLB5u+1T4Hs5CKD/PVWBJl+/sXRpxXS1zijllZWbDipcMBM")
value === returned_secret2 // false
```

Can this be related to `grpc-gateway` or stack configuration?

KrishnaIyer, 2021-06-03:
Ahhh of course.  I get it now. The value returned is the encrypted value and is hence garbage.

KrishnaIyer, 2021-06-03:
This is related to https://github.com/TheThingsIndustries/lorawan-stack-support/issues/405.

@bafonins and I discussed offline and decided that it's better to find a proper fix for both issues instead of a patch. I'll sync with @kschiffer for the UX and update the issue.

KrishnaIyer, 2021-06-08:
I had a discussion with @kschiffer and the easiest thing to be done here is to return the plaintext value of the LBS LNS Secret on the result of the `UpdateGateway` request. I can do this on the backend.

PR #4355 (2021-07-13): Return plaintext values for update gateway

#### Summary

Closes #4066 

#### Changes

- Return plaintext values of secrets set in `UpdateGateway`.
- Remove unused error message.

#### Testing

Local stack

[image]

##### Regressions

Don't expect any since I don't expect clients to use the encrypted value that this RPC used to return for anything.

#### Notes for Reviewers

@htdvisser If there's a prettier way to do this, let me know.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] C
closed 2021-07-13 by KrishnaIyer