TheThingsNetwork/lorawan-stack #4076: Include network identifiers in upstream metadata

johanstokking, 2021-04-20:
#### Summary

Include network identifiers in upstream metadata.

#### Why do we need this?

For subscribing networks to identify from which network the message originates. This is specifically useful for messaging based application-layer integrations, like webhooks.

#### What is already there? What do you see now?

Currently, when users build an HTTP endpoint to accept messages from the AS webhook integration, they have to include the originating network in the path, because it's not part of the metadata.

There is `X-Tts-Domain`, but that is intended for downlink callbacks. I would discourage using that for origin identification. Also, that is the request header and not the actual message, so if the accepting endpoint passes on the message as-is on an internal queue, this doesn't work well.

When receiving traffic that got routed via Packet Broker, the metadata includes `forwarder_net_id`, `forwarder_tenant_id` and `forwarder_cluster_id`. The community indicated that information is missing in metadata of gateways that are directly connected to the routing cluster that is handling the traffic.

#### What is missing? What do you want to see?

I would like to see an identification of the origin network in upstream messages.

I would like to see a network identification that follows LoRaWAN or LoRa Alliance best practices as much as possible. The main identifier is the NetID, which is already configured in the Network Server.

From LoRaWAN Backend Interfaces, there's also the "NSID"; and identifier of the Network Server within the namespace of the NetID. This is intended for different NS deployments that use the same NetID: effectively the "tenant ID" in The Things Stack.

Besides NetID and tenant ID, Packet Broker distinguishes client networks by cluster ID. This is also part of the Packet Broker Agent configuration already:

https://github.com/TheThingsNetwork/lorawan-stack/blob/b9e3712f32b748b31718896b3e83e153cdc55c6f/pkg/packetbrokeragent/config.go#L31-L33

> Packet Broker will combine tenant ID and cluster ID in the NSID (`tenant-id@cluster-id`) when it gets LoRaWAN Backend Interfaces 1.1 support.

So, I would suggest that we include `net_id`, `tenant_id` and `cluster_id` in upstream messages. This aligns directly with Packet Broker metadata and it provides an, in my view, necessary classification that we would not have with only NetID + NSID.

#### How do you propose to implement this?

Add `NetworkIdentifiers` message that gets added to `UplinkMessage`, set by NS based on the context, and AS copies that to `ApplicationUp`.

I'm not sure if NS already has an identifier of the cluster ID in the context or configuration, but this is needed anyhow for per-cluster tenant configuration in TTSE.

To decide is:

1. That we want this in the first place
2. That we use the terms `tenant_id` and `cluster_id`, and not a more opaque `ns_id` although that is more LoRaWAN Backend Interfaces 1.1 compliant
3. That we introduce a new identifiers message `NetworkIdentifiers`. I'm also fine with `DeploymentIdentifiers` or any other flavor. It wouldn't be related to entity identifiers.

#### How do you propose to test this?

Unit tests + some basic integration testing

#### Can you do this yourself and submit a Pull Request?

Can review

htdvisser, 2021-06-01:
I don't have anything to add here.

> To decide is:
> 
> 1. That we want this in the first place

ACK

> 2. That we use the terms `tenant_id` and `cluster_id`, and not a more opaque `ns_id` although that is more LoRaWAN Backend Interfaces 1.1 compliant

ACK

> 3. That we introduce a new identifiers message `NetworkIdentifiers`. I'm also fine with `DeploymentIdentifiers` or any other flavor. It wouldn't be related to entity identifiers.

I don't really have an opinion on this.

johanstokking, 2021-06-09:
OK thanks. Let's schedule this for 3.14.0 then.

@neoaggelos can you come up with a simple solution?

neoaggelos, 2021-06-21:
Network Server configuration already contains a `NetID` field: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/networkserver/config.go#L121

I suppose we could extend this config with a `ClusterID string`, which would identify the TTS instance. The Network Server then adds the NetID and ClusterID as metadata on uplink messages that are forwarded to the Application Server. For multi-tenant deployments, TenantID is also set (retrieved from the context).

I think this is simple enough to start. Two things that may need clarification:

- I assume Packet Broker verifies the `ClusterID` from the configuration, but for the Network Server it would be a purely informative field. For TTSC and TTSCE we can use well-known values, but people will be free to set this to whatever they want.
- Do we leave tenant_id empty for single-tenant deployments, or use a default value?

johanstokking, 2021-06-22:
> I suppose we could extend this config with a `ClusterID string`, which would identify the TTS instance. The Network Server then adds the NetID and ClusterID as metadata on uplink messages that are [...]

Sounds good.

> * I assume Packet Broker verifies the `ClusterID` from the configuration, but for the Network Server it would be a purely informative field. For TTSC and TTSCE we can use well-known values, but people will be free to set this to whatever they want.

Yes indeed. And this is a good point. We need to make sure that the cluster ID that NS uses is the same as what PBA uses to authenticate. Please take this into account when adding the configuration in https://github.com/TheThingsIndustries/lorawan-stack-aws (maybe one parameter?)

> * Do we leave tenant_id empty for single-tenant deployments, or use a default value?

Yes empty.

PR #4314 (2021-07-13): Include network identifiers in upstream messages

#### Summary

Closes #4076 

#### Changes

- Add new `NetworkIdentifiers` message.
- Set `NetworkIdentifers` when enqueueing `ApplicationUplink` messages to the Application Server.
- Add `ns.cluster-id` configuration option.
The changes follow the proposed implementation in the linked issue.

#### Testing

Locally

##### Regressions

The uplink message metadata may be getting too much with this change.

#### Notes for Reviewers

Draft until documentation and config PRs are done.

@johanstokking For the NetID, we use `types.NetID` with Gogoproto [... cut]
closed 2021-07-13 by neoaggelos