TheThingsNetwork/lorawan-stack #4077: Account owner can revoke right to change entities on their account

descartes, 2021-04-20:
#### Summary

From the forum: A user can change their own personal API key to turn off rights so they can't access their previously setup gateways/application etc

#### Steps to Reproduce

Under the user icon top right, go in to Personal API keys and tick boxes.

I haven't tried this myself but I coached the user in to getting an API key for his Data Storage integration running but it seems he's edited his own key whilst setting settings.

#### What do you see now?

Lots of active check boxes.

#### What do you want to see instead?

Lots of greyed out check boxes.

#### How do you propose to implement this?

Sorry, don't understand React enough to help out yet.

#### How do you propose to test this?

I'll setup some random accounts for testing things on my own copy of v3 so I'm not cluttering TTN v3.

#### Can you do this yourself and submit a Pull Request?
Nope

nejraselimovic, 2021-04-20:
@adriansmares assigning you because you encountered this issue already, but please reassign if someone else is more appropriate.

kschiffer, 2021-07-13:
Closing this since the console now renders a warning that should make users aware of this danger:

[image]

closed 2021-07-13 by kschiffer