TheThingsNetwork/lorawan-stack #4078: Warnings in the TTS Console for `gs.down.send` event

i3mSuperMan, 2021-04-20:
#### Summary
We are seeing a warning, i.e., `There was an error and events cannot be displayed and The raw data can be viewed by clicking this row` in the gateway live data of `TTSC` Console for the `gs.down.send` event.
#### Steps to Reproduce

1. Register an end-device and Gateway in TTSC Console.
2. Schedule a `downlink`.
3. You could see the `successfully schedule data downlink` event in the end-device i.e., `ns.down.data.schedule.success`
4. Navigate to the gateway live data, observe `gs.down.send` event is shown as a warning i.e., `There was an error and events cannot be displayed. The raw data can be viewed by clicking this row`

#### What do you see now?

- The warning is seen whenever a `downlink` is scheduled from the Network Server, but we can see the raw event by clicking on it (refer to the image).
- Observed across all TTSC clusters.
[image]
[image]
[image]

#### What do you want to see instead?
 `gs.down.send` event.
#### Environment
TTS v3.12.1
#### How do you propose to implement this?
Not sure
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No.

KrishnaIyer, 2021-04-21:
assigning @kschiffer  now so that he can assess/reassign.

johanstokking, 2021-04-21:
@i3mSuperMan what does the CLI report?

@KrishnaIyer this is very likely related to the `downlink_sent_total` metric not being incremented either.

i3mSuperMan, 2021-04-21:
@johanstokking 
The CLI events doesn’t seem to report the warning. They include `gs.down.send` event and everything seems fine.

mauimauer, 2021-04-24:
I am seeing the same (for downlink ACKs)

kschiffer, 2021-04-29:
This is a display logic issue in the Console. I'm working on a fix.

@i3mSuperMan can you post an example event payload (as text)?

i3mSuperMan, 2021-04-29:
@kschiffer, Here is an example of the `gs.down.send` event payload for the confirmed downlink
```
{
  "name": "gs.down.send",
  "time": "2021-04-29T10:14:05.406196093Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "",
        "eui": ""
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.DownlinkMessage",
    "raw_payload": "oGJQCCaAcgUBIGSB4tU=",
    "request": {
      "downlink_paths": [
        {
          "uplink_token": "ChsKGQoNdGVrdGVsaWMta29uYRIIZH/a//4Ae9QQ/OH4swwaDAjrjaqEBhCPxrzRAyDgsL3n9a0R"
        }
      ],
      "rx1_delay": 5,
      "rx1_data_rate_index": 5,
      "rx1_frequency": "868100000",
      "rx2_data_rate_index": 3,
      "rx2_frequency": "869525000",
      "frequency_plan_id": "EU_863_870_TTN"
    },
    "correlation_ids": [
      "as:downlink:01F4EHY37VFHS2NH33XWQZ7K8T",
      "gs:conn:01F4C94B13YZRGP62GET7TP2GR",
      "gs:up:host:01F4C94B1BM14V4Y6MER7VA470",
      "gs:uplink:01F4EHY2E8PRMKEQWCB2DV96M5",
      "ns:downlink:01F4EHY3TWB2HBDQ6YBMFV2TZ6",
      "ns:uplink:01F4EHY2F2SQ93K4KPGEHBQT7T",
      "rpc:/ttn.lorawan.v3.AppAs/DownlinkQueueReplace:5f808fba-45f1-44c9-a496-b7d3a516c20e",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01F4EHY2F2EQ9F0RKDWHSCK5RG",
      "gs:conn:01F4C94B13YZRGP62GET7TP2GR",
      "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F4EHY3TYKZV59V52SJEMVQMW"
    ]
  },
  "correlation_ids": [
    "gs:conn:01F4C94B13YZRGP62GET7TP2GR",
    "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F4EHY3TYKZV59V52SJEMVQMW"
  ],
  "origin": "ip-10-23-13-196.eu-west-1.compute.internal",
  "context": {
    "tenant-id": ""
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
  "unique_id": "01F4EHY3TYN2ESW7P5TKNFQ7T7"
} 
```

kschiffer, 2021-04-29:
Thank you. I'll file a PR that will fix this issue.

> I am seeing the same (for downlink ACKs)

Could you also post an example payload of such event? @mauimauer

If you do, make sure to redact sensitive data.

kschiffer, 2021-04-29:
Sorry, didn't mean to close this with #4101. I've merged a fix though for the original issue in the OP. Still waiting for more details about downlink ack messages.

mauimauer, 2021-04-29:
Here's where I get the warning
```
{
  "name": "gs.down.send",
  "time": "2021-04-29T13:51:59.289399292Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "redacted",
        "eui": "redacted"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.DownlinkMessage",
    "raw_payload": "YPpvCyaljgUDUv8AAfxVtm4=",
    "request": {
      "downlink_paths": [
        {
          "uplink_token": "Ch0KGwoPbWF6LXVrLWxkbi1hZzAxEgjcpjL//p+CbxDsi42SDBoMCP7zqoQGEOXi5qEDIODDpLbtvQs="
        }
      ],
      "rx1_delay": 5,
      "rx1_frequency": "867100000",
      "rx2_data_rate_index": 3,
      "rx2_frequency": "869525000",
      "priority": "HIGHEST",
      "frequency_plan_id": "EU_863_870_TTN"
    },
    "correlation_ids": [
      "gs:conn:01F4DE7T56CEFZTGY7H1AH4J7G",
      "gs:up:host:01F4DE7T5FK29DSTNXNJ7NMYF3",
      "gs:uplink:01F4EYD2WW77AX1CBX27ANCAND",
      "ns:downlink:01F4EYD39RSWFAP0B11K0ESWXY",
      "ns:uplink:01F4EYD2WXS46AN7BYYVQACWQQ",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01F4EYD2WXJZ1AZCJ0XBJD1ZHX",
      "gs:conn:01F4DE7T56CEFZTGY7H1AH4J7G",
      "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F4EYD39SVY6P7BRV7D3GR2NC"
    ]
  },
  "correlation_ids": [
    "gs:conn:01F4DE7T56CEFZTGY7H1AH4J7G",
    "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F4EYD39SVY6P7BRV7D3GR2NC"
  ],
  "origin": "ip-10-100-12-202.eu-west-1.compute.internal",
  "context": {
    "tenant-id": "redacted"
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
  "unique_id": "01F4EYD39SWT4HBWQNGEKVRGV2"
}
```

kschiffer, 2021-04-29:
Thank you very much @mauimauer.

I can confirm that my PR solves the display issue with this event as well. Note though that this is not a downlink ack event, which have the `name`: `gs.down.tx.success`.

PR #4101 (2021-04-29): Fix display of gateway downlink events in the Console
#### Summary
This PR will fix display issues of gateway downlink events.

References #4078 

#### Changes
- Distinguish gateway from end device downlink messages and use distinct display logic for gateways
- Fix wrong icon used for gateway downlink events
- Update event data used in event story

#### Testing

Manual testing using realistic data within storybook

##### Regressions

Shouldn't be any, but if then they would be related to previews for downlink messages in the Console event view.

#### Notes for Reviewers
This doesn't close #407 [... cut]
closed 2021-04-29 by kschiffer