TheThingsNetwork/lorawan-stack #4085: Transmit confirmation event for LBS gateways in Gateway live data

i3mSuperMan, 2021-04-22:
#### Summary
As per the earlier insights from the [#253](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/253#issuecomment-747599562),
RTT is only recorded on successful TxACK messages for downlinks (always for UDP and from v3.10.4 [for LBS](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.10/CHANGELOG.md#3104---2020-12-08) ).

We believe for the above to happen there must be a `gs.down.tx.success` for the LBS gateways right after `gs.down.send` upon the transmit confirmation. Ref: https://doc.sm.tc/station/tcproto.html#transmit-confirmation

#### Steps to Reproduce

1. Register an end-device and LBS based Gateway in TTS Console.
2. Schedule a `downlink`.
3. You could see the `successful scheduled data downlink` event in the end-device i.e., `ns.down.data.schedule.success`
4. Navigate to the gateway live data, we can only see `gs.down.send` event but no Ack message/event i.e., `gs.down.tx.success`

#### Why do we need this?
To understand the scheduled downlink status from the Gateways via the Gateway live data.

#### What is already there? What do you see now?
- For UDP gateways, there is the `gs.down.send` and `gs.down.tx.success` events upon successful downlink.
```
{
  "name": "gs.down.send",
  "time": "2021-04-22T13:36:23.973621056Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "",
        "eui": ""
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.DownlinkMessage",
    "raw_payload": "",
    "request": {
      "class": "CLASS_C",
      "downlink_paths": [
        {
          "uplink_token": "ChoKGAoMbXVsdGl0ZWNoLTAxEggAgAAAoAA2iBDs2pEQGgwIzPeFhAYQ+Pan+gIg4Nutin4="
        }
      ],
      "rx2_data_rate_index": 3,
      "rx2_frequency": "869525000",
      "frequency_plan_id": "EU_863_870_TTN"
    },
    "correlation_ids": [
      "as:downlink:01F3WWQGMH38HE64K4X33BF0CZ",
      "ns:downlink:01F3WWQGV6J6DVXDC68WP7SH2V",
      "rpc:/ttn.lorawan.v3.AppAs/DownlinkQueueReplace:55770fa5-a8b3-4d3a-9174-7e9673048594",
      "gs:conn:01F3WWP7CD9AMAM6ASA8K0HKDX",
      "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F3WWQGX59PEK02RCV8B0WVYK"
    ]
  },
  "correlation_ids": [
    "gs:conn:01F3WWP7CD9AMAM6ASA8K0HKDX",
    "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F3WWQGX59PEK02RCV8B0WVYK"
  ],
  "origin": "ip-10-23-13-196.eu-west-1.compute.internal",
  "context": {
    "tenant-id": ""
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
{
  "name": "gs.down.tx.success",
  "time": "2021-04-22T13:36:29.678398322Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "",
        "eui": ""
      }
    }
  ],
  "correlation_ids": [
    "gs:conn:01F3WWP7CD9AMAM6ASA8K0HKDX",
    "gs:tx_ack:01F3WWQPFE4KRY9Y0CZZXQX9TR"
  ],
  "origin": "ip-10-23-13-196.eu-west-1.compute.internal",
  "context": {
    "tenant-id": ""
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
  "unique_id": "01F3WWQPFE23D83DYTPE30YH3Y"
}
```
- In the LoRa Basics Station gateways, we can see a successful scheduled downlink in the console live data, but couldn’t see the `gs.down.tx.success` event in the gateway events.
```
{
  "name": "gs.down.send",
  "time": "2021-04-21T02:41:00.814069747Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "",
        "eui": ""
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.DownlinkMessage",
    "raw_payload": "",
    "request": {
      "class": "CLASS_C",
      "downlink_paths": [
        {
          "uplink_token": "ChkKFwoLdHRpZy0wMDIyMjISCFigy//+gAyxEPyfs+oBGgwIs6H+gwYQ1d3Q0AMg4OCxoKcO"
        }
      ],
      "rx2_data_rate_index": 3,
      "rx2_frequency": "869525000",
      "frequency_plan_id": "EU_863_870_TTN"
    },
    "correlation_ids": [
      "as:downlink:01F3S4TR5B0YT80DEQV3P70DA4",
      "ns:downlink:01F3S4TRC9YX4NJN18FSX9HX36",
      "rpc:/ttn.lorawan.v3.AppAs/DownlinkQueueReplace:d9a6c4ab-9429-4442-a3ff-d073348627ca",
      "gs:conn:01F3S4BE8GZDJV8ZY2Y5P6TMAY",
      "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F3S4TRC9W35N379V52CB7YF4"
    ]
  },
  "correlation_ids": [
    "gs:conn:01F3S4BE8GZDJV8ZY2Y5P6TMAY",
    "rpc:/ttn.lorawan.v3.NsGs/ScheduleDownlink:01F3S4TRC9W35N379V52CB7YF4"
  ],
  "origin": "ip-10-23-15-98.eu-west-1.compute.internal",
  "context": {
    "tenant-id": ""
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
  "unique_id": "01F3S4TRCERT7YPZPTFA84YWJS"
}
```
#### What do you want to see instead?
`gs.down.tx.success` event for LBS based gateway

#### Environment:
TTSC v3.12.1

#### How do you propose to implement this?
...

#### How do you propose to test this?
...

#### Can you do this yourself and submit a Pull Request?
No

KrishnaIyer, 2021-04-22:
@i3mSuperMan: In the LBS case, is the downlink successfully transmitted an do you see a `dntxed` message in the gateway logs for the downlink? 
We do emit Tx ACK messages for LBS as well (actually even before we supported that for UDP). So if you're not receiving this event, then there's a bug somewhere.

If you have a local test setup, can you try to enable debug logs and look for  the `Tx acknowledgment either does not correspond to a downlink message or arrived too late` warning message?

KrishnaIyer, 2021-04-28:
Moving this to the next milestone as there are still missing details and today is the deadline for `v3.12.2.`.

i3mSuperMan, 2021-05-13:
Hi @KrishnaIyer,

We have tested with Tektelic Kona Micro, TTIG and Multitech Gateways using LBS on TTSC v3.12.2.

**Tektelic Kona Micro with LBS:**

- For the successful downlinks, we are seeing `gs.down.send` & `gs.down.tx.success` in the Gateway live data.
- From the Gateway packet forwarder logs, we see that the `dnmsg` and `dntxed` in the Gateway. Refer below.
```
May 13 05:19:45 kona-micro tek_bstn[1899]: [DEBUG ] bridge_bstn_lns.c:433 payload:
{ "DR": 5, "Freq": 868500000, "upinfo": { "rctx": 0, "rxtime": 1620883185, "xtime": 756927220, "gpstime": 0, "rssi": -35, "snr": 9.0, "chan": 7 }, "msgtype": "updf", "MHdr": 64, "DevAddr": 638094548, "FCtrl": 0, "FCnt": 24, "FOpts": "", "FPort": 1, "FRMPayload": "d3", "MIC": 604007012 }
May 13 05:19:47 kona-micro tek_bstn[1899]: [DEBUG ] bridge_bstn_lns.c:373 data received from WS stream:
{"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":14600,"pdu":"a0d48c08268003007c4809351064f0","RxDelay":1,"RX1DR":5,"RX1Freq":868500000,"priority":25,"xtime":760927220,"rctx":0,"MuxTime":1620883186.9415317}
May 13 05:19:47 kona-micro tek_bstn[1899]: [DEBUG ] bridge_data_conv.c:266 sending to: 127.0.0.1:45885
May 13 05:19:47 kona-micro tek_bstn[1899]: [DEBUG ] bridge_data_conv.c:266 sending to: 127.0.0.1:45885
May 13 05:19:47 kona-micro tek_bstn[1899]: [DEBUG ] bridge_bstn_lns.c:432 sending [144] bytes to WS stream
May 13 05:19:47 kona-micro tek_bstn[1899]: [DEBUG ] bridge_bstn_lns.c:433 payload:
{ "msgtype": "dntxed", "DevEui": "00-00-00-00-00-00-00-00", "rctx": 72057594038112900, "xtime": 760927220, "diid": 14600, "txtime": 1620883187 }
May 13 05:19:57 kona-micro tek_bstn[1899]: [DEBUG ] bridge_data_conv.c:266 sending to: 127.0.0.1:45885
```
**The Things Indoor Gateway:**

- We can see a successful scheduled downlink (`gs.down.send`) in the Gateway live data, but **couldn’t** see the `gs.down.tx.success` event in the gateway events.
- From the Gateway packet forwarder logs, we see that the `dntxed` in the Gateway. Refer below.

```
2021-05-13 06:01:26.650 [S2E:VERB] RX 867.1MHz DR5 SF7/BW125 snr=9.2 rssi=-71 xtime=0x9900001CD93183 - updf mhdr=40 DevAddr=26088CD4 FCtrl=80 FCnt=180 FOpts=[] 01EB mic=-1342237280 (14 bytes)
2021-05-13 06:01:27.249 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:28.253 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:29.257 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:30.261 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:31.265 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:32.270 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:33.274 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:34.278 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:34.979 [S2E:DEBU] ::0 diid=39552 [ant#0] - next TX start ahead by 321ms33us
2021-05-13 06:01:35.291 [S2E:VERB] ::0 diid=39552 [ant#0] - starting TX in 19ms886us
2021-05-13 06:01:35.296 [SYS:DEBU]   Free Heap: 18792 (min=16792) wifi=5 mh=7 cups=8 tc=4
2021-05-13 06:01:35.306 [S2E:INFO] TX ::0 diid=39552 [ant#0] - dntxed: 869.525MHz 27.0dBm ant#0(0) DR3 SF9/BW125 frame=60D48C08268011000CD9A92A12BA0D
2021-05-13 06:01:35.465 [S2E:DEBU] Tx done diid=39552
```
**Multitech with LBS:**

- We can see a successful scheduled downlink (`gs.down.send`) in the Gateway live data, but **couldn’t** see the `gs.down.tx.success` event in the gateway events.
- From the Gateway packet forwarder logs, we see that the `dnmsg` in the Gateway. Refer below.
```
2021-05-11 15:21:39.810 [AIO:XDEB] [3|WS] > {"msgtype":"updf","MHdr":64,"DevAddr":33614848,"FCtrl":128,"FCnt":44,"FOpts":"","FPort":1,"FRMPayload":"EB","MIC":-1279818904,"RefTime":1620746499.809954,"DR":5,"Freq":867100000,"upinfo":{"rctx":0,"xtime":16607023628964651,"gpstime":0,"fts":-1,"rssi":-65,"snr":8.5,"rxtime":1620746499.809976}}
2021-05-11 15:21:40.357 [AIO:XDEB] [3|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","dC":0,"diid":15,"pdu":"6000ec00028508000351ff00033fa0bf8b","RxDelay":1,"RX1DR":5,"RX1Freq":867100000,"priority":25,"xtime":16607027927931947,"rctx":0,"MuxTime":1620746500.3681977}
2021-05-11 15:21:40.358 [S2E:WARN] ::0 diid=15 [ant#0] - Tx job too far ahead: 1h11m
```
So, from the above tests, for Tektelic the stack emits the Tx Ack message but not for TTIG, Multitech. Unfortunately, we couldn’t successfully connect Multitech to local setup.

KrishnaIyer, 2021-05-14:
Thanks @i3mSuperMan. I'll need some time to look into this so I'm moving it to the next milestone.

closed 2021-07-13 by KrishnaIyer