TheThingsNetwork/lorawan-stack #4087: Profile settings link is broken in non-EU TTN clusters

megabug, 2021-04-23:
#### Summary

The profile settings link (added by #3968) is broken in non-EU TTN clusters.

#### Steps to Reproduce

1. Log in to a non-EU TTN cluster, i.e. https://au1.cloud.thethings.network/console/
2. Click on username in top right, click "Profile settings"
3. Observe that you are redirected to https://au1.cloud.thethings.network/oauth/profile-settings, which DNE

#### What is already there? What do you see now?

Redirect to https://au1.cloud.thethings.network/oauth/profile-settings

#### What is missing? What do you want to see?

Redirect to correct URL (I'm assuming it's https://eu1.cloud.thethings.network/oauth/profile-settings)

#### Environment

v3.12.1

#### How do you propose to implement this?

Add new config item for absolute URL to profile settings page?

#### Can you do this yourself and submit a Pull Request?

Nope 😅

nejraselimovic, 2021-04-28:
@asmulko please assign @michalborkowski96 if needed.

kschiffer, 2021-04-28:
Thanks for filing this issue @megabug 

This is an issue with our deployment configuration for TTN, which we will update shortly. I'll close this issue since it is not directly related to TTS istelf.

closed 2021-04-28 by kschiffer