TheThingsNetwork/lorawan-stack #4090: Allow configuring the MaxEIRP as part of device MAC settings

neoaggelos, 2021-04-23:
#### Summary

Allow configuring MaxEIRP for devices as part of the MAC settings

#### Why do we need this?

Currently, MaxEIRP can only be set in `MACState.DesiredParameters`. This is OK for ABP devices, but impossible for OTAA devices, since MAC state can only be configured if the device already has a session.

#### What is already there? What do you see now?

MaxEIRP is set in `MACState.DesiredParameters`. When initializing the MAC state of a device, the Network Server will use the default MaxEIRP setting from the respective band (https://github.com/TheThingsNetwork/lorawan-stack/blob/0e95225c3688a6d42e4f3bd2463f38258863bb91/pkg/networkserver/mac/utils.go#L221-L228).

This is also related to the Device Repository, where `maxEIRP` can be configured per device profile.

#### What is missing? What do you want to see?

Ability to configure MaxEIRP for OTAA devices.

#### Environment

3.12.1

#### How do you propose to implement this?

Introduce `MACSettings.MaxEIRP`. Network Server respects that value (if set) when initializing a new MAC state.

#### How do you propose to test this?

Unit tests, creating devices from the device repository

#### Can you do this yourself and submit a Pull Request?

Yes, but I am not sure if this is the correct way to do it. Adding to 3.12.2 because this needs to be addressed along with #4086

neoaggelos, 2021-04-28:
@johanstokking Please comment on whether the suggested implementation makes sense. I am not really familiar with how the Network Server uses this value, and if this something that should be configured per device.

johanstokking, 2021-04-28:
Yes indeed, the MaxEIRP is definitely a device capability. This should be in the MAC state on reset.

PR #4128 (2021-05-19): Add MACSettings.DesiredMaxEIRP field

#### Summary

Closes #4090 

#### Changes

- Introduce `DeviceEIRPValue` struct (containing an enum value of type `DeviceEIRP`)
- Introduce `MACSettings.DesiredMaxEirp` field.
- Consider MaxEirp value for `mac.DeviceDesiredMaxEIRP`
- Miscellaneous changes required for validations and setting the DesiredMaxEirp field from the CLI.

#### Testing

- Unit tests
- Setting `desired_max_eirp` field from the CLI and verify that the `DesiredParameters.MaxEIRP` value is set properly after device joins.

##### Regressions

There should not be any, the default value f [... cut]
closed 2021-05-19 by neoaggelos