TheThingsNetwork/lorawan-stack #4091: ttn-lw-cli exits 0 with invalid subcommand

jchonig, 2021-04-23:
#### Summary
ttn-lw-cli should exit with a non-zero exit code with an invalid subcommand

...

#### Steps to Reproduce
See log below

#### What do you see now?
```
jch@jch-mbp[531]:~$ ttn-lw-cli user foo; echo $?
User commands

Usage:
  ttn-lw-cli users [command]

Aliases:
  users, user, usr, u

Available Commands:
  api-keys           Manage user API keys
  contact-info       Manage user contact info
  create             Create a user
  create-login-token Create a user login token
  delete             Delete a user
  forgot-password    Request a temporary user password
  get                Get a user
  invitations        Manage user invitations
  list               List users
  oauth              Manage OAuth authorizations and access tokens
  purge              Purge a user
  restore            Restore a user
  rights             List the rights to a user
  search             Search for users
  set                Set properties of a user
  update-password    Update a user password

Flags:
  -h, --help   help for users

Global Flags:
      --allow-unknown-hosts                             Allow sending credentials to unknown hosts
      --application-server-enabled                      Application Server enabled (default true)
      --application-server-grpc-address string          Application Server address (default "localhost:8884")
      --ca string                                       CA certificate file
  -c, --config strings                                  Location of the config files (default [.ttn-lw-cli.yml,/Users/jch/.ttn-lw-cli.yml,/Users/jch/Library/Application Support/.ttn-lw-cli.yml])
      --credentials-id string                           Credentials ID (if using multiple configurations)
      --device-claiming-server-grpc-address string      Device Claiming Server address (default "localhost:8884")
      --device-template-converter-grpc-address string   Device Template Converter address (default "localhost:8884")
      --dump-requests                                   When log level is set to debug, also dump request payload as JSON
      --gateway-server-enabled                          Gateway Server enabled (default true)
      --gateway-server-grpc-address string              Gateway Server address (default "localhost:8884")
      --identity-server-grpc-address string             Identity Server address (default "localhost:8884")
      --input-format string                             Input format (default "json")
      --insecure                                        Connect without TLS
      --join-server-enabled                             Join Server enabled (default true)
      --join-server-grpc-address string                 Join Server address (default "localhost:8884")
      --log.level string                                The minimum level log messages must have to be shown (default "info")
      --network-server-enabled                          Network Server enabled (default true)
      --network-server-grpc-address string              Network Server address (default "localhost:8884")
      --oauth-server-address string                     OAuth Server address (default "https://localhost/oauth")
      --output-format string                            Output format (default "json")
      --packet-broker-agent-grpc-address string         Packet Broker Agent address (default "localhost:8884")
      --qr-code-generator-grpc-address string           QR Code Generator address (default "localhost:8884")

Use "ttn-lw-cli users [command] --help" for more information about a command.
0
jch@jch-mbp[532]:~$ ttn-lw-cli gateway foo; echo $?
Gateway commands

Usage:
  ttn-lw-cli gateways [command]

Aliases:
  gateways, gateway, gtw, g

Available Commands:
  api-keys             Manage gateway API keys
  claim                Claim a gateway (EXPERIMENTAL)
  collaborators        Manage gateway collaborators
  contact-info         Manage gateway contact info
  create               Create a gateway
  delete               Delete a gateway
  get                  Get a gateway
  get-connection-stats Get connection stats for a gateway
  list                 List gateways
  list-frequency-plans List available frequency plans for gateways
  purge                Purge a gateway
  restore              Restore a gateway
  rights               List the rights to a gateway
  search               Search for gateways
  set                  Set properties of a gateway

Flags:
  -h, --help   help for gateways

Global Flags:
      --allow-unknown-hosts                             Allow sending credentials to unknown hosts
      --application-server-enabled                      Application Server enabled (default true)
      --application-server-grpc-address string          Application Server address (default "localhost:8884")
      --ca string                                       CA certificate file
  -c, --config strings                                  Location of the config files (default [.ttn-lw-cli.yml,/Users/jch/.ttn-lw-cli.yml,/Users/jch/Library/Application Support/.ttn-lw-cli.yml])
      --credentials-id string                           Credentials ID (if using multiple configurations)
      --device-claiming-server-grpc-address string      Device Claiming Server address (default "localhost:8884")
      --device-template-converter-grpc-address string   Device Template Converter address (default "localhost:8884")
      --dump-requests                                   When log level is set to debug, also dump request payload as JSON
      --gateway-server-enabled                          Gateway Server enabled (default true)
      --gateway-server-grpc-address string              Gateway Server address (default "localhost:8884")
      --identity-server-grpc-address string             Identity Server address (default "localhost:8884")
      --input-format string                             Input format (default "json")
      --insecure                                        Connect without TLS
      --join-server-enabled                             Join Server enabled (default true)
      --join-server-grpc-address string                 Join Server address (default "localhost:8884")
      --log.level string                                The minimum level log messages must have to be shown (default "info")
      --network-server-enabled                          Network Server enabled (default true)
      --network-server-grpc-address string              Network Server address (default "localhost:8884")
      --oauth-server-address string                     OAuth Server address (default "https://localhost/oauth")
      --output-format string                            Output format (default "json")
      --packet-broker-agent-grpc-address string         Packet Broker Agent address (default "localhost:8884")
      --qr-code-generator-grpc-address string           QR Code Generator address (default "localhost:8884")

Use "ttn-lw-cli gateways [command] --help" for more information about a command.
0
```

...

#### What do you want to see instead?

A non-zero exit code as in:

```
jch@jch-mbp[540]:~$ ttn-lw-cli foo; echo $?
unknown command "foo" for "ttn-lw-cli"
255
```

...

#### Environment
```
jch@jch-mbp[536]:~$ brew --version
Homebrew 3.1.3-67-gafbe0e8
Homebrew/homebrew-core (git revision e997d9f5a0; last commit 2021-04-23)
jch@jch-mbp[537]:~$ ttn-lw-cli version
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.12.1
Build date:          2021-04-15T18:40:21Z
Git commit:          ca69664da
Go version:          go1.16.3
OS/Arch:             darwin/amd64
```
...

#### How do you propose to implement this?

...

#### How do you propose to test this?

See logs above

...

#### Can you do this yourself and submit a Pull Request?

I don't have any golang experience, yet.

...

htdvisser, 2021-04-28:
This is related to an issue in our `cobra` CLI library: https://github.com/spf13/cobra/issues/1156, so you may want to read that to see how we can fix it.

neoaggelos, 2021-05-05:
The Cobra issue has been stale for a long time, and [a PR that fixes this issue in kind](https://github.com/kubernetes-sigs/kind/pull/2090/files) opted for setting `RunE` in all top-level commands and returning an error for invalid subcommand there.

cc @htdvisser, Is this acceptable in our case? I have started some work in https://github.com/TheThingsNetwork/lorawan-stack/tree/issue/4091-error-on-invalid-subcommand following this approach.

closed 2021-05-11 by neoaggelos