TheThingsNetwork/lorawan-stack #4094: Error when send downlink to device with Class A and Class C.

viethoa14, 2021-04-28:
#### Summary

Now when we use cellular network for TTN, we saw sometime error when NS send downlink to end-device on class A but with class C, we saw many error. Show log on gateway, we saw error report "Packet REJECTED, timestamp seems wrong, too much in advance"
...

#### Steps to Reproduce

1. Make "Schedule downlink" to end-device class C
2. Show log on "console server"
3. No response from end-device to NS  

#### What do you see now?

this is log when we show on Kerlink gateway 

Apr 26 07:41:45 Wirnet local1.info lorad[870]: <6> Sent 1 uplink message
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> Received uplink message:
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> | lora uplink (39C0001F), payload 22 B, channel 921.4 MHz, crc ok, bw 125 kHz, sf 12, cr 4/5
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> | Unconfirmed Data Up, DevAddr FC005049, FCtrl [ADR], FCnt 279, FPort 2
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> |  - radio (00000000)
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> |   - demodulator counter 1133303204, UTC time 2021-04-26T07:41:45.1359Z, rssi -83 dB, snr 3.75< 8.75 <13.5 dB
Apr 26 07:41:45 Wirnet local1.info lorafwd[865]: <6> Uplink message (AF7E) sent
Apr 26 07:41:47 Wirnet local1.info lorafwd[865]: <6> Uplink message (AF7E) acknowledged in 2240.17 ms
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> Downlink message (000F) received
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> Received downlink message:
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> | lora downlink (0000000F), payload 17 B, required 1, preamble 8 B, header enabled, crc disabled, polarity inverted
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> | Unconfirmed Data Down, DevAddr FC005049, FCtrl [ADR], FCnt 109
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> |  - radio (00000000), channel 921.4 MHz, bw 125 kHz, sf 10, cr 4/5, power 20 dB
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> |   - transmission (00000000), priority 1, on counter 1135303204
Apr 26 07:41:48 Wirnet local1.info lorad[870]: <6> Received downlink message
Apr 26 07:41:48 Wirnet local1.err lorad[870]: <3> Packet REJECTED, timestamp seems wrong, too much in advance (current=1136222185, packet=1135303204, type=0)
Apr 26 07:41:48 Wirnet local1.err lorad[870]: <3> Failed to enqueue downlink message
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> Received uplink message: transmission event (0000000F / 00000000), status "Bad timing"
Apr 26 07:41:48 Wirnet local1.info lorafwd[865]: <6> Downlink message (000F) acknowledged
Apr 26 07:41:51 Wirnet local1.info lorafwd[865]: <6> Heartbeat (AFB9) sent
Apr 26 07:41:51 Wirnet local1.info lorafwd[865]: <6> Heartbeat (AFB9) acknowledged in 82.7868 ms
Apr 26 07:41:52 Wirnet local1.info lorafwd[865]: <6> Uplink message (AF7F) sent
Apr 26 07:41:52 Wirnet local1.info lorafwd[865]: <6> Uplink message (AF7F) acknowledged in 83.4784 ms
Apr 26 07:42:01 Wirnet local1.info lorafwd[865]: <6> Heartbeat (AFBA) sent
Apr 26 07:42:01 Wirnet local1.info lorafwd[865]: <6> Heartbeat (AFBA) acknowledged in 90.0331 ms

#### What do you want to see instead?

...

#### Environment

We test  with Lorawan stack 3.11.3 -> Lorawan stack 3.12.1
Gateway: Kerlink Wirnet Station, ver 3.6, cpf 1.1.6. We tried enable or disable "Schedule downlink late" 
Device: version MAC 1.0.2, regional parameters version PHY V1.0.2 REV A

#### How do you propose to implement this?

1. We can not know problem with Lorawan'config or not. When we use other provider with same infrastructure, this issue is disappear.
2. We must change parameter Lorawan'config, Gateway'config, clock ...or not  

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2021-04-28:
> Packet REJECTED, timestamp seems wrong, too much in advance (current=3138816049, packet=3138729107, type=0)

Looking at this difference, expressed in microseconds, the packet is 0.086942 ms too late. I'm afraid this is caused by spikes in the latency: up to 2.4 seconds in your excerpt of the ping output.

Indeed, the "schedule late" functionality is designed for this. By default, this is 800 milliseconds. For your backhaul, it should be somewhere closer to 2000 milliseconds. Now, this does mean that there is less time for the network and the application layer to generate downlink messages. You need an RX1 delay of at least 5 seconds (which is the default in The Things Stack).

closed 2021-04-28 by johanstokking

viethoa14, 2021-04-29:
Thanks for your support!
When packet don't send, delay for backhaul around 100ms but when we send downlink or end-device send uplink, sometime delay will increase.  
When delay 100ms, we still see error when schedule downlink.
(config for schedule:  
gs:
   ...........
  udp:
    rate-limiting:
      enable: 'true'
      messages: '10'
      threshold: '10ms'
    schedule-late-time: '2000ms'
)

Beside we see warning on server'log: "stack_1      | redis: 2021/04/29 10:38:43 commands.go:36: specified duration is 789ms, but minimal supported value is 1s - truncating to 1s" , now we are using redis server version 6.0.12 (Redis version=6.0.12, bits=64, commit=00000000, modified=0, pid=1, just started)
Is there any connection between the two problems?

johanstokking, 2021-04-29:
> When packet don't send, delay for backhaul around 100ms but when we send downlink or end-device send uplink, sometime delay will increase.
> When delay 100ms, we still see error when schedule downlink.

The delay with cellular backhaul isn't constant. It isn't related to sending uplink and downlink messages. It's inherent to how 3G works.

> Beside we see warning on server'log: "stack_1 | redis: 2021/04/29 10:38:43 commands.go:36: specified duration is 789ms, but minimal supported value is 1s - truncating to 1s" , now we are using redis [...]

Yes, but that's ok.